pathfinder-for-autonomous-navigation/FlightSoftware #54: Teensy Compiler Flags for Floating Point Math

nhz2, 2019-11-06:
This issue is a place to discuss what compiler flags relating to floating point math we want to use for flight software. [Default compiler flags for the Arduino framework on PIO.](https://github.com/platformio/platform-teensy/blob/master/builder/frameworks/arduino.py)

nhz2, 2019-11-06:
Previously I used -fno-math-errno to get the teensy to use the assembly instruction for `sqrtf` instead of the actual function.
build_unflags = -fmath-errno
build_flags =  -fno-math-errno

nhz2, 2019-11-06:
Also -fsingle-precision-constant should be not be defined, because otherwise any double literal without an `L` suffix will automatically be cast to single precision. Generally it's good practice to add the `L` suffix if you want a double literal, and an `f` suffix for single precision literals, but there might be some libraries we are using that don't do this.

build_unflags = -fsingle-precision-constant

kkrol27, 2019-11-06:
@nhz2 Do you know if the `long double` type the same as `double` on our ARM chips?

tanishqaggarwal, 2019-11-06:
I don't know much about this, but I'll follow along on the discussion and go with whatever ends up being the best idea.

kkrol27, 2019-11-06:
As mentioned in slack, marking everything as a `long double` can cause template argument deductions issues. See below:

    include/gnc_ode.hpp:241:5: note:   template argument deduction/substitution failed:
    test/ode_test/test.cpp:105:78: note:   deduced conflicting types for parameter 'T' ('double' and 'long double')
   code = gnc::ode23(ti, 7.5L, yi, yf, ne, bf, 1e-4L, 1e-6L, 1e-6L, 2000, fsho);

Due to this I'd like to pursue more of a compiler flag fix than marking things as a `long double`.

kkrol27, 2019-11-06:
This may also need to be expanded to a general compiler flags discussion. We are setting a good amount of extra stuff that we don't technically even have control over. Here are the default sets of compiler flags defined by PlatformIO for the Teensy35:

    teensy35.build.command.gcc=arm-none-eabi-gcc
    teensy35.build.command.g++=arm-none-eabi-g++
    teensy35.build.command.ar=arm-none-eabi-gcc-ar
    teensy35.build.command.objcopy=arm-none-eabi-objcopy
    teensy35.build.command.objdump=arm-none-eabi-objdump
    teensy35.build.command.linker=arm-none-eabi-gcc
    teensy35.build.command.size=arm-none-eabi-size
    teensy35.build.flags.common=-g -Wall -ffunction-sections -fdata-sections -nostdlib
    teensy35.build.flags.dep=-MMD
    teensy35.build.flags.optimize=-Os
    teensy35.build.flags.cpu=-mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant
    teensy35.build.flags.defs=-D__MK64FX512__ -DTEENSYDUINO=147
    teensy35.build.flags.cpp=-fno-exceptions -fpermissive -felide-constructors -std=gnu++14 -Wno-error=narrowing -fno-rtti
    teensy35.build.flags.c=
    teensy35.build.flags.S=-x assembler-with-cpp
    teensy35.build.flags.ld=-Wl,--gc-sections,--relax,--defsym=__rtc_localtime={extra.time.local} "-T{build.core.path}/mk64fx512.ld" -lstdc++
    teensy35.build.flags.libs=-larm_cortexM4lf_math -lm

nhz2, 2019-11-07:
Would just casting the constants to `double` after adding the `L` solve that issue? On 32 bit arm, `long double` is a synonym for `double`

shihaocao, 2019-12-04:
Pasting Kyle's comments from the AttitudeEstimator control task here for reference:

> The interface with the GNC algorithm looks great and I see absolutely no issues there. Couple things to note however,
> 
> * Compiler flags for `double`/`float` constant treatments need to be double checked otherwise we could see precision issues in the FSW build only (which would be a headache to debug).
> * `*const` or `&` treatment of state registry input values could potentially be looked at.

shihaocao, 2019-12-04:
> Also, the following flags must be included for teensy builds to ensure that 1.0e9 is treated as a double and not a float:
> 
> ```
> build_unflags = -fsingle-precision-constant -fmath-errno
> build_flags = -fno-math-errno
> ```
> 
> Not sure if that has trickled into FSW yet.

^more comments

tanishqaggarwal, 2020-01-18:
Making this a bug since we need to address the compiler flags issue.

nhz2, 2020-01-20:
Here is a link to the optimization flags for teensy. [https://docs.platformio.org/en/latest/platforms/teensy.html](https://docs.platformio.org/en/latest/platforms/teensy.html)

tanishqaggarwal, 2020-01-22:
Closing since we've fixed this shit in #158

closed 2020-01-22 by tanishqaggarwal