pathfinder-for-autonomous-navigation/FlightSoftware #69: ADCS Driver Questions

nhz2, 2019-11-14:
What are the units of the magnetorquer and wheel commands in the ADCS driver?

Also, how should the exponential filters coefficients be set to ignore all history?

kkrol27, 2019-11-14:
Am^2 and torque in MKS.

kkrol27, 2019-11-14:
On the adcs side at least that’s what the fixed point conversion is done with. However cuz it’s just a scale factor FSW can make it whatever you want (think @shihaocao used the same constants however so the units should be the same).

kkrol27, 2019-11-14:
The exponential filter constants should be left to there defaults and you should be good to go.

nhz2, 2019-11-14:
Is the speed command in rad/s ?

kkrol27, 2019-11-14:
Oh forgot about that one, yep should be

tanishqaggarwal, 2019-12-09:
Is this item closed?

closed 2020-01-22 by tanishqaggarwal