pathfinder-for-autonomous-navigation/FlightSoftware #92: Pan Epoch Internal State Field Creation

shihaocao, 2019-12-03:
We/Shihao need/needs to figure out where (which control task) to create/define the Pan Epoch Internal State Field because the state field is necessary for gnc/attitude estimation calculations. The Pan Epoch Internal Statefield should be a gps_time_t.

nhz2, 2019-12-03:
The actual value of Pan Epoch is in psim, in `include/gnc_constants.hpp` its called `gnc::constant::init_gps_week_number` to convert to nanoseconds, just cast to a int64_t and then multiply by (7L * 24L * 60L * 60L * 1000L * 1000L * 1000L), 
@kkrol27  maybe the nanoseconds version should be another constant in psim?

shihaocao, 2019-12-04:
Resolved by defining pan_epoch is not a state field at all, and instead, as a constant pulled from gnc_constants

closed 2019-12-04 by shihaocao