pathfinder-for-autonomous-navigation/FlightSoftware #98: Piksi upgrades + test

tanishqaggarwal, 2019-12-08:
- Take Quake SBDIX into account
- Testing Piksi dead
- Test Piksi float -> fixed RTK

tanishqaggarwal, 2019-12-08:
Need to modify ClockManager as well to update "propagated" GPS time

tanishqaggarwal, 2019-12-17:
Need to unit-test that the output values are initialized to NaNs

tanishqaggarwal, 2019-12-30:
The reported GPS position should be the GPS position of the _docking port face_ (this is distinct from the GPS position of the GPS antenna face.) A position offset therefore needs to be added for all CDGPS measurements.

tanishqaggarwal, 2019-12-30:
We need to add a time offset since the hardware-stored GPS time offset is sometime in 2000

tanishqaggarwal, 2019-12-30:
> The reported GPS position should be the GPS position of the _center of mass_ (this is distinct from the GPS position of the GPS antenna face.) A position offset therefore needs to be added for all CDGPS measurements.

@nhz2 @kkrol27 In my notes from last semester I found that we want to offset the computed GPS position for the sake of the control algorithm. Is this still necessary under the new orbital control algorithm?

kkrol27, 2020-02-19:
> @nhz2 @kkrol27 In my notes from last semester I found that we want to offset the computed GPS position for the sake of the control algorithm. Is this still necessary under the new orbital control [...]

This will really only be an issue when we have RTK and are in very close approach.

shihaocao, 2020-09-20:
PIksi should set values to NaN anytime an uncessusful read was performed that control cycle

tanishqaggarwal, 2020-11-19:
Piksi data gets reported into state fields in mm, not m

kylekrol, 2021-03-16:
We're only listening to fixed RTK mode, NaNs are a thing of the past and taken of somewhere else. The only potentially open issue is Piksi readings during SBDIX; however, we're deeming this very low priority.