pathfinder-for-autonomous-navigation/FlightSoftware #123: Determine the number of control cycles needed in QuakeManager

athena255, 2019-12-26:
Right now the max_config_cycles, max_wait_cycles, etc (in QuakeManager.h) have placeholder values. 

In hardware tests, SBDWB, SBDRB, CONFIG do not take any longer than one control cycle. 

SBDIX takes anywhere from 70 to 210 control cycles (but averages ~120).

tanishqaggarwal, 2019-12-29:
Can you update the "Recurring Constants Review" spreadsheet on the OneDrive? I think that's a good centralized place to handle constants issues like this one.

tanishqaggarwal, 2019-12-29:
https://cornellprod-my.sharepoint.com/:x:/r/personal/saa243_cornell_edu/_layouts/15/Doc.aspx?sourcedoc=%7B10E398A9-3D68-44F8-BE1B-E4FA5DBECBC8%7D&file=Recurring%20Constants%20Review.xlsx&action=default&mobileredirect=true

closed 2019-12-30 by athena255