pathfinder-for-autonomous-navigation/FlightSoftware #125: Nuanced Fault Response Upgrades

shihaocao, 2019-12-27:
I think the Fault class should have a writeable boolean state field, say `unsignal_f,` that will trigger the `unsignal()` command, then automatically reset back to false. I'm worried about the following cases:

1. A fault flag becomes tripped.
2. On the ground, we deduce what caused it, and also realize a set of commands to upload to prevent it from occurring in the future.
3. We upload those commands, and merely ask the satellite to unsignal(), that way if somehow if it happens again, we can be notified.

Or slightly differently:

1. A fault flag becomes tripped.
2. We deduce what caused it, and realized there's nothing we can do to prevent it from being tripped again (say it'll happen every few days), but there is a set of commands to properly recover from it.
3. We upload the recovery commands, as well as the unsignal() command so that the satellite can continue to function until the fault condition occurs again (and still protect itself from the fault).

Further, in regard to `num_consecutive_faults` being able to still increment when a suppression or override is active, I think the release of a suppression or override should not modify `num_consecutive_faults`. If we do want `num_consecutive_faults` to be reset, merely follow the release with an `unsignal_f = true`.

Cases where I think this behavior is useful:

1. num_consecutive_faults == 0 for some fault initially
2. we suppress that fault because even if it is flagged we don't want it to be triggered in a critical window
3. we carry out the critical task, during which maybe the fault condition happens, or maybe it doesn't
4. when we unsuppressed, we don't follow it immediately with an `unsignal()`, we want the fault to immediately flag if the condition was met during the suppression period

tanishqaggarwal, 2020-01-30:
Let's also make persistence ground-configurable.

shihaocao, 2020-01-31:
@tanishqaggarwal 

In order to actually call `unsignal()` when `unsignal_f` is true, I think the best place to put that is in MissionManager.

Basically MissionManager should have a block of code that first parses all the `unsignal_f`'s, and calls `unsignal()` on all faults before running `check_hardware_faults`

tanishqaggarwal, 2020-01-31:
This sounds like a good idea. I've been thinking of implementing a separate fault controllers class though, to prevent code bloat inside mission manager.

tanishqaggarwal, 2020-02-01:
Being implemented on #185

closed 2020-02-02 by tanishqaggarwal