pathfinder-for-autonomous-navigation/FlightSoftware #134: I'm worried this might be physically impossible, or at least lead to issues:

shihaocao, 2020-01-06:
I'm worried this might be physically impossible, or at least lead to issues:

Since both satellites are physically identical, it's impossible for both Satellites to be docked and have both +X faces facing Zenith.

**EDIT:** If the satellites are docked, the +X faces will only be 90 degrees apart from one another.

Ideas:

- Have both satellites +X faces 45 degrees offset from one another and approach optimally for docking but not for comms
- Have both +X target Zenith, we accept that they will approach not optimally set for docking, and then do something drastic once we get close.

Additional thoughts? Stewart suggests making sure these kinds of fine-tuning what angles we want to approach at be commandable in case we realize one might be more ideal than one another (perhaps to adjust comms performance).

_Originally posted by @shihaocao in https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/pull/120/files/ca4f9a623ba9f69455e6baa041bb9333143844c7_

nhz2, 2020-01-06:
Won't the satellite's +X faces have to be 180 degrees apart when they are docked? I'm assuming the docking face on both leader and follower are the same, is one rotated, or could it be rotated 180 degrees?

shihaocao, 2020-01-06:
@nhz2 
Since the "axis" of the actual blue docking piece is 45 degrees offset, it causes the X-axis to only be 90 degrees apart.

See this helpful diagram that Stewart drew:
https://prnt.sc/qk4iwu

Imagine both satellites are water bottles on tables, now flip across the dotted line as if you were turning one of the water bottles upside down. The docking face dimples will be perfectly aligned but the +X faces will be 90 degrees offset.

nhz2, 2020-01-06:
Oh, that makes sense.

shihaocao, 2020-01-18:
Prelim conclusion:
Choose option 1
> Have both satellites +X faces 45 degrees offset from one another and approach optimally for docking but not for iridium comms

To implement: Use offsets similarly for optimal solar, merely point a long edge towards zenith instead of the iridium patch location

kkrol27, 2020-04-10:
Closing in favor of #287.

closed 2020-04-10 by kkrol27