pathfinder-for-autonomous-navigation/FlightSoftware #139: Magnetometer calibration task

tanishqaggarwal, 2020-01-17:
> Right now I am calibrating the magnetometer after eclipse, and after detumbling.

_Originally posted by @nhz2 in https://github.com/pathfinder-for-autonomous-navigation/psim/pull/92_

This needs to get implemented into FSW.

shihaocao, 2020-04-23:
FSW:

- Monitor needs to set registers
- Controller needs timer to set modes into calibrate,
- Put a statefield to control timer, and offsets, independent for both

ADCS:

- Adjust readings based off of offset
- Implement calibrate for mag1
- Actually call calibrate in imu setup,
- And whenever mode is in calibrate

tanishqaggarwal, 2020-09-02:
Sending to the backlog since #424 tracks the meat of the work. #424 is the science required prior to implementing this ticket.

kylekrol, 2021-08-07:
Covered by #701 and #702.

closed 2021-08-07 by kylekrol