pathfinder-for-autonomous-navigation/FlightSoftware #174: Let's be more careful about Quake error codes

tanishqaggarwal, 2020-01-29:
A lot of the error codes don't indicate a _hardware_ failure, but rather bad luck with comms. These bad luck cases shouldn't cause the Quake fault response, which is listed in #20 .

tanishqaggarwal, 2020-04-10:
Part of this issue is also making sure the radio error codes are being downlinked.

athena255, 2020-05-08:
Quake manager will downlink the values in radio.err whose values are given by ``Devices::radio_status_t`` enum located in ``Qlocate.hpp``. If you are referring to the error codes such as 18, 38, those can be added to the QuakeManager. However, since the the ``radio.err`` state field only holds a single value, we will only know what the latest fault was.

athena255, 2020-05-08:
There's also an ambiguity issue which is that since the radio error codes now use enums (0 to 9), their values conflict with the MO status values, which ranges from 0 to 36. So if we are unlucky, and it happens that we get an SBDIX response under 9 then we will not be able to tell whether the error is an MO status value or whether it is one of our implementation defined values.

tanishqaggarwal, 2020-05-08:
C++ enums allow you to specify an integer value for each item, which could mitigate this problem.

Perhaps we can add events for the radio data. That'll ensure we can capture multiple error codes, and then we'll also be able to capture when an event occurred.

athena255, 2020-09-20:
Quake will set the radio_err_f field upon experiencing any sort of error. However, there is no component other than QuakeManager that does anything with this field. Upon receiving an unexpected error, the Quake Manager will execute the config instructions.

athena255, 2020-09-20:
If handle_err is called, then Quake will transition to the wait (IDLE) state, wait 5 min, and then transition to config.

handle_err() is called in dispatch_transceive() in two places
1. transcieve ran out of cycles
In this case, the current behavior is that if transceive failed to reach Iridium, then we will try again as long as we have more than 1/3 of the maximum transceive cycles left. The correctness of this relies on the assumption that transceive is guaranteed to finish within 1/3 of the maximum transceive cycles. Currently max_transceive_cycles is (arbitrarily) set to 500. 

2. qct.execute(radio_state_t::transceive) returned an error
In this case, an error is only ever returned if the message is in the incorrect format. The specific control flow is as follows:

QuakeManager::dispatch_transceive()
---> int err_code = qct.execute(radio_state_t::transceive);
    QuakeControlTask::execute()
    --->dispatch_sbdix()
          --->quake.get_sbdix()
                QLocate::get_sbdix()
                 --->return parse_ints(buf + 8, sbdix_r);
                       parse_ints():
                       ---> returns OK as long as message in the expected format

Conclusions:
I'm not sure that we can guarantee that any given transceive state will finish in 1/3 (500) cycles. I think the move would be to modify dispatch_transceive() to transition directly to wait.

tanishqaggarwal, 2020-09-23:
Discussed offline the following points:
- I neglected to mention in my original specification of this ticket that we should think about MO status codes, not our own hardware-specific error codes. My concern was that gating the fault response on the condition that the MO status > 4 is a bad proposition, since, I thought that it would be better for the fault response to only be triggered by a failure in our hardware to contact the Iridium network, and not a failure in Iridium network internals. There are certain MO status codes that signal hardware failure and certain ones that signal Iridium network failure.

I now realize that (a) in the interest of descoping needless work and complexity, it's probably best to not make this distinction on MO status codes, and (b) it actually seems advantageous to try the fault response (powercycling, safehold, reboot) regardless of whose fault the comms failure was. At least it shakes up the spacecraft so that comms might be possible. My only concern would be if this causes frequent resets of the mission.

- It might not be a good idea to have dispatch_transceive() transition directly to `write` after comms failure, since if we've had a comms failure then there's an unfinished I/O transaction on the Quake's serial line, and it would not be wise to start another SBDWB transaction without clearing out the serial buffers. Going to `wait` prevents this by causing the QuakeManager to eventually transition to `config`, which (as far as we know) does a clean job of resetting buffers.

Closing this ticket since this ticket is therefore generating no additional software changes. @athena255 feel free to reopen if you disagree?

closed 2020-09-23 by tanishqaggarwal