pathfinder-for-autonomous-navigation/FlightSoftware #184: PropulsionSystem TimedLock implementation isn't overflow safe.

nhz2, 2020-02-01:
[https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/blob/master/src/fsw/FCCode/Drivers/PropulsionSystem.hpp](https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/blob/master/src/fsw/FCCode/Drivers/PropulsionSystem.hpp)

Using `micros` might prevent the lock from unlocking if `micros` wraps around. Some possible solutions to this are. 
1. Use an event [example](https://forum.pjrc.com/threads/44723-Arduino-Events/page3) to have the `TimedLock` auto unlock themselves after a certain number of milliseconds, either on the systick interrupt, or the next time `yield()`  is called.
2. Make a long int version of micros or nanos that won't ever overflow.

tanishqaggarwal, 2020-02-01:
@athena255

tanishqaggarwal, 2020-02-02:
Resolved by completely removing locking. #189

closed 2020-02-02 by tanishqaggarwal