pathfinder-for-autonomous-navigation/FlightSoftware #219: Prop planner

tanishqaggarwal, 2020-02-19:


tanishqaggarwal, 2020-04-16:
From our conversation today. Do you agree with this @nhz2 ?

Inputs:
an enum: start pressurizing, stop pressurizing, no command
On times for all four valves
Mission state

Outputs:
cycles_until_firing -> to prop controller
on_times -> to prop controller

Behavior:
Perform the following in the follower and follower_close_approach states:
- If start_pressurizing is true, set cycles_until_firing to be greater than min_cycles_needed(). Send through the orbit controller’s commanded on_time to the prop controller
- If stop_pressurizing is true, set cycles_until_firing to 0 and send orbit controller’s commanded on_times to the prop controller
- If no_command is true, release control of cycles_until_firing and on_time commands to the prop controller.

nhz2, 2020-04-16:
### Here's an example GNC output and the expected physical response.
 
Let me know what you think.

Notes |   | GNC commands |   |   |   |   | Expected Physical Response (delayed by no more than a few control cycles) |   |   |   |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
  | Control Cycle | pressurize enum | on_time 1(ms) | on_time 2(ms) | on_time 3(ms) | on_time 4(ms) |   | expected impulse 1(s) | expected impulse 2(s) | expected impulse 3(s) | expected impulse 4(s)
  | 0 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 1 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 2 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 3 | start | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 4 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 5 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | ... | ... | ... | ... | ... | ... |   | ... | ... | ... |  
  | 3000 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 3001 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
GNC will wait some time after start before commanding a non zero on_time | 3002 | no command | 100 | 20 | 100 | 0 |   | 0.1 | 0.02 | 0.1 | 0
  | 3003 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | ... | ... | ... | ... | ... | ... |   | ... | ... | ... |  
  | 7000 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7001 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7002 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
GNC will wait some time after sending the last non zero on_time before sending another | 7003 | no command | 0 | 300 | 50 | 700 |   | 0 | 0.3 | 0.05 | 0.7
  | 7004 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7005 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
GNC can command to stop pressurizing when it won't fire again for a long time | 7006 | stop | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7007 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7008 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7009 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7010 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
GNC can start pressurizing and then stop without commanding a non zero on_time | 7011 | start | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7012 | stop | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7013 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7014 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7015 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
GNC can start pressurizing and then stop without commanding a non zero on_time | 7016 | start | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7017 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7018 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7019 | stop | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7020 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7021 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0
  | 7022 | no command | 0 | 0 | 0 | 0 |   | 0 | 0 | 0 | 0

tanishqaggarwal, 2020-05-03:
After discussion today, behavior changes as follows:

Behavior:
Perform the following in the follower and follower_close_approach states:

If start_pressurizing is true, set cycles_until_firing to be greater than min_cycles_needed(). Send through the orbit controller’s commanded on_time to the prop controller
If stop_pressurizing is true, release control of cycles_until_firing and on_time commands to the prop controller.
If FIRE is true (this is a renaming of no_command), set cycles_until_firing to 0 and send orbit controller’s commanded on_times immedaitely to the prop controller.

tanishqaggarwal, 2020-10-30:
This is obsoleted by work on the Orbit control task.

closed 2020-10-30 by tanishqaggarwal