pathfinder-for-autonomous-navigation/FlightSoftware #234: Gomspace Stops Charging

shihaocao, 2020-03-01:
The gomspace stops charging once gomspace controller writes settings to the GomSpace. When you comment out the code that writes settings to the gomspace, meaning only read data requests are present, FSW no longer stops GomSpace from charging.

shihaocao, 2020-04-09:
Discussion from testing on 04/08:

In flight we should be defaulting to PPT auto.
Further, we should only be calling `gs.set_pv_volt()` when `ppt_mode.get() == 2` where 2 is fixed ppt.

Unanswed q's:
You can't charge gomspace with power supply in auto PPT mode or else whack stuff happens. How to handle this?
Gomspace default values how?

shihaocao, 2020-04-09:
Find default settings on gomspace

shihaocao, 2020-04-15:
Gomspace should default to Auto PPT in flight, but on the ground, fixed is fine, unsure how to let it charge tho

stewartaslan, 2020-05-08:
This will be fixed by #386, so it's redundant.

tanishqaggarwal, 2020-05-08:
Determined to be a symptom of #386: the PPT mode is not actually writable by the Gomspace driver, for some reason.

tanishqaggarwal, 2020-09-02:
Closing and tracking on #485

closed 2020-09-02 by tanishqaggarwal