pathfinder-for-autonomous-navigation/FlightSoftware #255: Ptest upgrades

shihaocao, 2020-03-08:
- Have Ptest automatically call cycle as an option
- Have Ptest take *args
- Support for multiple arg types

tanishqaggarwal, 2020-03-09:
See addition of cycle() option on the `prop_driver_case` branch

tanishqaggarwal, 2020-03-09:
Could you clarify what you mean by *args?

shihaocao, 2020-03-12:
@kkrol27 Do you mind if you actually clarify? I'm actually not too familiar with optional arguments in python.

kkrol27, 2020-03-12:
Here I was mainly talking about better support for vectors and potentially look into supporting "native" Python types as opposed to always having to pass strings to things (see #251). Changes would mainly affect override and write state and would be backwards compatible.

For example, when writing a vector to a state instead of doing this:

    self.sim.flight_controller.write_state("adcs_cmd.rwa_speed_cmd", "0, 0, 0")

we could make this possible:

   self.sim.flight_controller.write_state("adcs_cmd.rwa_speed_cmd", 0.0, 0.0, 0.0)

We could add native support for types like booleans:

    self.sim.flight_controller.write_state("dcdc.ADCSMotor_cmd", True)

instead of:

    self.sim.flight_controller.write_state("dcdc.ADCSMotor_cmd", "true")

These changes would be pretty easy to implement by moving to a `*args` and `**kwargs` paradigm. I'm totally happy to tackle this tomorrow/this weekend if it sounds like good changes! Should probably get more involved in ptest anyway.

PR #327 (2020-03-27): Cleaner Write State Syntax for Vectors and Bools in PTest
# Cleaner Write State Syntax for Vector and Bools in PTest

Fixes #255.

### Summary of changes

Slightly reworked the write state interface to allow for:

* Feeding a variable number of arguments to the write state function to specify a vector:

        sim.write_state('some.vector.state.field', 0.0, 0.0, 0.0)

* Passing native booleans to the write state function as opposed to strings:

        sim.write_state('some.boolean.state.field', True)

* The `timeout` argument is now passed as a keyword argument and still default to `None`.

 [... cut]
closed 2020-03-27 by kkrol27