pathfinder-for-autonomous-navigation/FlightSoftware #287: Integrate GNC into ADCSCommander

shihaocao, 2020-03-12:
Place GNC functions into ADCSCommander

kkrol27, 2020-03-12:
@shihaocao could you expand on which GNC functions we are referencing here?

shihaocao, 2020-03-12:
I've actually got no idea what the names of them are but,
They are the functions that convert the desired vectors to actual MTR and Wheel commands.

kkrol27, 2020-03-12:
Ahhh okay so stuff on top of the controller itself. We can talk about this this weekend and find the companion PSim tickets.

kkrol27, 2020-03-12:
Just assigning myself too so I don't forget.

kkrol27, 2020-03-28:
@shihaocao After talking with @tanishqaggarwal and @stewartaslan I'm just going to take this over for next week to take some stuff off your plate.

kkrol27, 2020-03-29:
Per @nhz2 request, we'll also need to update documentation here in accordance with implementation: https://pan-software.readthedocs.io/en/latest/flight_software/components.html

kkrol27, 2020-04-10:
The current pointing strategy defined in FSW needs to be reworked have the long edge of the satellite along the orbit normal and point the antenna phase outward at some angle `phi` specified in the hill frame.

PR #517 (2020-10-25): Attitude Controller
# Attitude Controller

Fixes #287 

### Summary of changes
- Removed AttitudeComputer from MCL
- Added AttitudeController (does not need temporary data structures for mtr and rwa_torques)
- Added AttitudeController pio unit test
- Bump psim and lin submodule pointers
- ADCSCommander no longer needs to read pointing vectors, instead reads suggested mtr and rwa_torques
- AttitudeComputer replaced by AttitudeController in FSW, MCL
- ADCSCommander is restructured to hopefully be simpler
- ADCSCommander has upgraded pio unit test
- New differentialation of adcs_state's:
- `point [... cut]
closed 2020-10-25 by shihaocao