pathfinder-for-autonomous-navigation/FlightSoftware #326: Create Interface for Handling Editable GNC Constants

kkrol27, 2020-03-27:
There are a ton of constants that GNC uses internally for it's various algorithms. One good example are the gains used by the attitude PD controller (PR for that is [here](https://github.com/pathfinder-for-autonomous-navigation/psim/pull/158)).

We want FSW to be be able to treat these values as state fields so they can be set from the ground and managed by FSW (for example when docking the moment of inertia and controller gains should change).

My initial thought here is it may be easiest to make a "GNC" type state field. A templated class that behaves exactly like a state field except the `get` and `set` functions interact with an extern variable not held in the state field class itself. Assuming the state field class hierarchy can support this, we'd also need to make `lin` objects serializable. Otherwise stuff could get messy.

@tanishqaggarwal and anyone else, thoughts would be appreciated!

kkrol27, 2020-03-27:
I refrained from adding it to the schedule for the time being until we've scoped out the work involved here.

nhz2, 2020-03-27:
Editable constants is an oxymoron. But yeah, I agree that there are a lot of GNC parameters that we might want to adjust from ground. Maybe there should be a control task or class that owns all of the shared GNC parameters as state fields, and the other GNC control tasks would take them in as input state fields.

nhz2, 2020-03-27:
Also won't non constant globals break simulating the leader and follower in MATLAB because both sats would share the same globals?

kkrol27, 2020-03-27:
> Also won't non constant globals break simulating the leader and follower in MATLAB because both sats would share the same globals?

If you are talking about HOOTLs with flight software, no because they will be running separate binaries. If we're talking about running sims when calling cxx code, I'm not 100% sure how the MEX files work, as in can you store persistent state information in cxx outside of MATLAB?

Also, we can rename from constants if we really want to - it's definitely a little goofy.

nhz2, 2020-03-27:
Yeah I think a mex function will have persistent global variables between calls. But why not just use the state field registry instead of having global variables?

kkrol27, 2020-03-27:
> Yeah I think a mex function will have persistent global variables between calls. But why not just use the state field registry instead of having global variables?

Because we aren't going to have access to that code within PSim. To get things up and running I was just getting together something that had low overhead and we could easily plug in to the state field registry on the FSW side of things.

nhz2, 2020-03-27:
Ok that makes sense. I think though it would be cleaner if things like moment of inertia were in the estimator state, and the pd gains were in the pointing controller state, even if there is a bit of overhead, and it requires changes to those structs. Because just making things that are constants not constant by using global variables seems like a short cut that might create more problems in the future.

tanishqaggarwal, 2020-03-27:
I've started some work on lin-based serializers on the branch `lin_statefields`.

kkrol27, 2020-03-27:
What data types do we think we'll end up supporting in terms of sizes? I know supporting any randomly sized vector or matrix would be pretty tough. If we are going for that though let me know and I could help explain some of lin's template meta-programming nonsense.

tanishqaggarwal, 2020-04-10:
Oops I was late to this issue, but currently flight software only supports 3-4 element lin fields.

tanishqaggarwal, 2020-04-10:
We really just need the GNC state field in order to close out this issue. Discussed during software tagup today