pathfinder-for-autonomous-navigation/FlightSoftware #350: Add State Field for "PAN Time" to the Clock Manager

kkrol27, 2020-03-30:
We need to add a state field to provide time in seconds/nanoseconds since the "PAN epoch". This will be fed to many of the GNC algorithms and environmental models.

shihaocao, 2020-09-27:
Still need this:

- Note that Piksi GPS time isn't a reading at "now" but rather a data point that's been in the buffer for a while
- Need a Kalman filter on the GPS time
- Increase uncertainty when GPS data isn't coming in, noise from Piksi
- If data becomes too uncertain (exceed a uncertainty threshold) or if we regress too far forward in time or behind when we suddenly get a new data point, this could be cause to either:
-- Reset everything,
-- Or apply a small delta t spread over many cycles to catch up, (wondering if this might mess up other estimations...)

tanishqaggarwal, 2021-01-03:
Might as well make this part of the orbit estimator.

Note: shihao's comment above is out of date.

PR #446 (2020-05-01): Update the Attitude Estimator to the UKF Interface
# Update the Attitude Estimator to the UKF Interface

Closes #348.
Closes #350.
Closes #445.

### Summary of changes
- Added the `orbit.time` state field in the `FieldCreatorTask` which holds the time in seconds since the PAN epoch as a double.
- Update the `AttitudeEstimator` interface:
  - Added the `attitude_estimator.mag_flag` state field to chose a default magnetometer sensor.
  - Added the `attitude_estimator.w_body` state field which is the current angular rate estimate.
  - Removed the `attitude_estimator.h_body` state field.
  - Added [... cut]
closed 2021-01-03 by tanishqaggarwal