pathfinder-for-autonomous-navigation/FlightSoftware #354: ADCS HAVT - GYR and SSA ADC#

shihaocao, 2020-03-31:
Sometimes, IMU GYR and ALL SSA ADC's report as non-functional.
More debugging is required.

shihaocao, 2020-03-31:
Highly suspect this is because the MAG2 is down. Next steps: replace mag 2, and replicate behavior

shihaocao, 2020-03-31:
Behavior to replicate: If either gomspace or the adcs_monitor or adcs_controller CT's are executing (and you cycle from ptest), the GYR and SSA ADC'S go down (seen in ADCS debug HAVT heartbeat)

shihaocao, 2020-03-31:
We think there could be hardware EMI from Gomspace I2C and or FC to ADCSC I2C to Wire 2 on ADCSC (between all the SSA ADCS and IMU GYR)

stewartaslan, 2020-04-03:
Cut the SCL and SDA traces to Mag 2 ([as-run procedure](https://cornellprod-my.sharepoint.com/:w:/r/personal/saa243_cornell_edu/_layouts/15/Doc.aspx?sourcedoc=%7B3c4c8faf-eeda-41be-a620-886a812627f8%7D&action=edit&wdPid=56966d08)). Everything passes on the ETU sat now.

[image]

closed 2020-04-03 by stewartaslan