pathfinder-for-autonomous-navigation/FlightSoftware #378: HOOTL slows down over long periods of time

tanishqaggarwal, 2020-04-08:
See the attached graph of pan.cycle_no, which increments almost logarithmically. Need to investigate this.
[image]

shihaocao, 2020-04-08:
Do you have the data points for this? Was thinking about checking it on log log plot

kkrol27, 2020-04-08:
I'd look for some operations that `O(log n)` or `O(n log n)` where somehow `n` is directly related to number of cycle lmao This is pretty funny.

tanishqaggarwal, 2020-04-08:
I think it came from here:
https://cornellprod-my.sharepoint.com/personal/saa243_cornell_edu/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Fsaa243%5Fcornell%5Fedu%2FDocuments%2FOAAN%20Team%20Folder%2FSubsystems%2FSoftware%2FPull%20Request%20HITL%20Logs%2F%23359%20%2D%20Docking%20ptest%20checkout%20and%20DockingSystem%2BDockingController%20fixes%2FDockingCheckoutCase%5F20200331%2D155335

kkrol27, 2020-04-09:
As in the log writing? How exactly is that handled?

tanishqaggarwal, 2020-04-20:
I think I discovered the issue. TinyDB, the database that backs the incoming telemetry, is slow:

[image]

This test runs 100,000 iterations of writing 100 values to the database.

Over time, as the database gets large, the insertion operation to the database has time complexity <img src="https://render.githubusercontent.com/render/math?math=O(n)">. Hence, the plot of time per control cycle vs control cycle # should be proportional to <img src="https://render.githubusercontent.com/render/math?math=n^2">, which is exactly the behavior we see.

I'll open a PR to add this stress test to the database, but I think we can consider this issue closed (unless we want to fix this issue somehow.)

PR #412 (2020-04-20): Datastore stress test (which solved #378)
Fixed #378.

### Summary of changes
- Added a datastore stress test that reports a linear increase in insertion time. Running the stress test requires setting the `STRESS_TEST` environment variable:

```
export STRESS_TEST=1
python -m pytest ptest/tests/datastore_stress_test.py
```

Results:
[image]

### Testing
This is a PR that adds testing.

### Constants
No constants added.

### Telemetry
No telemetry added.

### Documentation Evidence
Documentation with the issue tickets is sufficient, since this is just a bug, not a feature.
closed 2020-04-20 by tanishqaggarwal