pathfinder-for-autonomous-navigation/FlightSoftware #383: assert(false) still fails on HITLs

tanishqaggarwal, 2020-04-09:
I noticed that assert(false) causes a hang on Teensy HITLs but not HOOTLs when the size of a flow was too large in a first draft of #371. This concerns me, because it's exactly the wrong behavior, and may persist through flight? We need to take a more careful look at the build flags.

tanishqaggarwal, 2020-05-08:
We might have to put a custom assert into GNC, and then use it in FSW.

athena255, 2020-05-08:
Adding ``${teensy.build_flags}`` to  line 15 to get rid of assert statements. fsw_native_leader does not call the assert statements either.

athena255, 2020-05-08:
[image]
[image]

closed 2020-05-12 by athena255