pathfinder-for-autonomous-navigation/FlightSoftware #386: Set hardcoded constants in the Gomspace control task, rather than the initially read values

tanishqaggarwal, 2020-04-10:
From GomspaceController.cpp
```
if (control_cycle_count==1){
        power_cycle_output1_cmd_f.set(false);
        power_cycle_output2_cmd_f.set(false);
        power_cycle_output3_cmd_f.set(false);
        power_cycle_output4_cmd_f.set(false);
        power_cycle_output5_cmd_f.set(false);
        power_cycle_output6_cmd_f.set(false);

        pv1_output_cmd_f.set(gs.hk->vboost[0]);
        pv2_output_cmd_f.set(gs.hk->vboost[1]);
        pv3_output_cmd_f.set(gs.hk->vboost[2]);

        ppt_mode_cmd_f.set(gs.hk->pptmode);

        heater_cmd_f.set(gs.get_heater());

        counter_reset_cmd_f.set(false);
        gs_reset_cmd_f.set(false);
        gs_reboot_cmd_f.set(false);
    }
```

This is a problem, since it's dynamically setting some pretty important values. Instead, change these values to be tracked constants so that the effect of flight software on hardware is well-defined.

tanishqaggarwal, 2020-04-10:
These values may also have to be added to EEPROM-saving.

tanishqaggarwal, 2020-04-17:
We need to potentially modify the checkout case since we haven't done extensive HITL testing with it yet.

tanishqaggarwal, 2020-09-02:
Closing and tracking on #485

closed 2020-09-02 by tanishqaggarwal

fatimayousuf, 2021-02-04:
@tanishqaggarwal @kylekrol @shihaocao what should the default value for the heater command be?