pathfinder-for-autonomous-navigation/FlightSoftware #409: Gyro heater integration into ADCS Software and Flight Software

tanishqaggarwal, 2020-04-19:
This never got done lmao

shihaocao, 2020-04-23:
remove, p i d fields

kkrol27, 2020-04-23:
Here is an old PR for reference: https://github.com/pathfinder-for-autonomous-navigation/ADCSSoftware/pull/15

* We're going implement a bang bang controller so we need to remove the `gyr_temp_p`, `gyr_temp_i`, and `gyr_temp_d` registers.
* Add a driver class for the heater and put another in device in the HAVT.
* Implement a bang bang controller in `imu.cpp`

shihaocao, 2020-04-27:
[image]

weirdly low temps on the gyro_temp, investigating

shihaocao, 2020-04-27:
above was performed on a minimal hitl

shihaocao, 2020-05-06:
lol that took a lot more work than it was supposed to xd

PR #440 (2020-05-06): Added Gyro Heater
# Added Gyro Heater

Fixes #409 

### Summary of changes
- Added GyroHeater to FSW and ADCSC
- Removed 2 statefields, renamed 1, added 1 set
- Added GyroHeaterDiag(nostic)Case - used for manually verifying heater working
- Small refactoring of ADCSCheckoutCase

GyroHeater is disabled by default. Needs to be enabled by FSW (change?). Default PWM strength of 255 (max), and a target of 20C.

### Testing
Passed on EDU sat HITL, 255 pwm, target 40C, held to within 40.2 and 39.6C

Logs in HITL Logs

### Constants
N/A

### Telemetry
flow_data.csv modified.

new fields: adcs_monitor. [... cut]
closed 2020-05-06 by shihaocao