pathfinder-for-autonomous-navigation/FlightSoftware #422: If pressure sensor fails, then PropController will not be able to fire at all

athena255, 2020-04-23:
The current implementation in PropController relies on the pressure sensor. If the pressure sensor fails, then the PropController will not be able to tell when it should stop pressurizing. The state machine within the PropController requires that the pressure sensor report that the pressure in the outertank (Tank2) is above the threshold firing pressure (25 psi). If this is not the case, then the PropController will not be allowed to enter the Firing state.

tanishqaggarwal, 2020-04-24:
The main concern behind this ticket is that this leaves us with no mechanism for manually controlling valves from the ground in the event that we want to investigate the issue from the ground. One proposed solution is to create a "safe manual" mode for PropController, which is like PropController's current manual mode, but actually enforces a 2 ms delay between valve openings and closes them after some specified duration. Thoughts @stewartaslan @kkrol27 ?

tanishqaggarwal, 2020-04-25:
Implementation decision: instead of checking the pressure, we check the pressure _fault_. That way, the behavior to block entry to `firing` can be cleanly suppressed.

PR #465 (2020-05-12): Prop faults
# Propulsion Autonomous Fault Handling

Fixes #422,  #295, #218, #375

### Summary of changes

PropulsionSystem.cpp
- get_fake_temp_analog() fixes a bug where the the prop driver was using the a pointer owned by the parent class instead of the derived class. 

MainFaultHandler.cpp
- changed PropOverpressureFaultHandler to PropFaultHandler

PropController.hpp
- definitions for PropState_Venting
- ActionCycleOpenClose class has been created to factor out the common behavior used by pressurizing and venting
- num_prop_firings_f keeps track of the number of firings that have  occurred. [... cut]
closed 2020-05-12 by athena255