pathfinder-for-autonomous-navigation/FlightSoftware #439: Update files on remote server

fatimayousuf, 2020-04-28:
The remote server files need to be updated. 
- There shouldn't be a `main.py` file anymore (running `git pull` didn't change any of the files in gsw for me). 
- All the config files that are on the the .gitignore also need to be updated. 
- There is no tlm or MCT folders as of now. Again, running `git pull` didn't change anything for me.
- There exists both `/PAN/src/FlightSoftware/src/gsw` and `PAN/gsw`? We shouldn't need both...

I'm thinking maybe just re-clone the entire repository???

tanishqaggarwal, 2020-04-28:
Recloning seems like the right thing to do. Go ahead.

The PAN/gsw is actually a symlink into PAN/FlightSoftware/src/gsw

fatimayousuf, 2020-04-29:
Everything works! 😊

closed 2020-04-29 by fatimayousuf