pathfinder-for-autonomous-navigation/FlightSoftware #450: Standardize state types

athena255, 2020-05-02:
State types of Control Tasks have been declared as unsigned int, unsigned char, char, etc. This causes issues with testing because `WritableStateField<unsigned char>*`  is not the same as `WritableStateField<unsigned int>*`. 

Idea: Typedef unsigned char as `STATE_T` and make all Control Task states of type STATE_T.

kkrol27, 2020-05-02:
Is this only an issue with control tasks that don't return `void`? Would be easier to just remove the template argument for `ControlTask` and force all of them to return an integer code or char code even if it doesn't mean anything?

tanishqaggarwal, 2020-10-30:
Almost certainly not going to have time to implement this.

closed 2020-10-30 by tanishqaggarwal