pathfinder-for-autonomous-navigation/FlightSoftware #485: Gomspace driver issues

tanishqaggarwal, 2020-09-02:
Major issues exist with the Gomspace driver implementation and/or hardware configuration settings that prevent it from passing PTest cases and exhibiting desirable behavior.

Reference tickets:
#234 
#386

tanishqaggarwal, 2020-10-01:
After testing today, we evaluated:
- A debug_console error causes counter values to be reported incorrectly.
- The hard reset command works as expected. Reboot doesn't seem to do anything, but hard reset does what we want it to (reset the entire spacecraft), so reboot doesn't matter.
- Unsure why PV voltages still don't change upon application. The plan is to see if the settings are correct on the flight units; if they are, we don't touch this function. If not, we have to scope out additional work to change the Gomspace Controller so that it can read/write config settings. We have to do this because the Gomspace USB terminal is not available on from the flight units.

tanishqaggarwal, 2020-10-04:
#506 addressed most driver issues except for the PV voltage bug. We suspect possible ESD issues with the EDU gomspace due to a history of being poorly treated.

Paths forward: test with another Gomspace, test with flight units (dangerous). If the flight units don't permit the behavior either it may be time to look into expanding the GomspaceController to be able to write config values to the Gomspace.

shihaocao, 2021-02-11:
The path forward is to just set GomSpace to default auto on the ground. We're going to ignore PV voltage setting problems.

For the best safety, satellites shall be charged when they are off.

18kudlugim, 2021-02-20:
[image]

18kudlugim, 2021-02-20:
Power cycling seems broken during gomspace hitl test. Is checking that all gomspace output channels are off at the same time an issue here?

kylekrol, 2021-04-01:
Closed by #647.

closed 2021-04-01 by kylekrol