pathfinder-for-autonomous-navigation/FlightSoftware #487: ADCS State Change x DCDC power state

shihaocao, 2020-09-04:
Sending the satellite into safehold will currently set the ADCS to enter `adcs::startup`. This causes the DCDC for ADCS to turn off. It is unknown what the effects of turning off and on the DCDC are with regard to ADCS performance.

There should be further testing done to validate functionality. Special care should be taken to validate motor performance since there have been issues in the past with motors failing to spin if the DCDC power was not supplied before sending motor commands.

If necessary, additional code might be required to reboot DCDC power to ADCS a full cycle before the ADCS state itself changes.

shihaocao, 2020-09-22:
Small note, this is actually distinct from #495. Since #495 refers to a full reboot, and #487 (this one) refers to a mission state change to a `safehold` and back out to `standby` vs ADCS functionality.

tanishqaggarwal, 2020-12-15:
Do we ever actually need to turn the DCDC off? Does it actually draw a lot of power? There's a lot of careful DCDC condition handling that we're doing right now that we should get rid of if possible.

knk38, 2021-03-03:
I think this issue should be closed as it is resolved by the SafeholdStandbyTransition case which verifies that mission manager does the correct state changes after throwing a fault and going into safehold. The new case was merged with PR #631

closed 2021-03-03 by knk38