pathfinder-for-autonomous-navigation/FlightSoftware #489: Powercyclings of devices need to be aware of Gomspace slow bootup time

tanishqaggarwal, 2020-09-06:
When turning a device on and off, the Gomspace takes about 30 seconds to slowly ramp down its voltage to zero. This is an issue because automated power cycling in the Quake Fault Handler assumes that powercycling can occur instantaneously. After #479 is merged, take care to address this in the code.

Also, currently there's no condition handling for if a power cycle fails. Should we consider turning this into a fault flag?

tanishqaggarwal, 2020-09-07:
After a closer look at the code I don't actually think this is an issue, because the Quake Fault Handler merely requests a powercycle and doesn't care what happens next. It has no dependence on whether or not the powercycle completed.

In the PTest case for the Quake fault handler, we also only check that the powercycle flag went high, and I feel like that's a sufficient check.

@shihaocao thoughts?

shihaocao, 2020-09-14:
Sorry, I didn't get a timely response in

I think your logic makes sense, however

Currently we have this on master:
`
    def check_quake_powercycled(self):
        if not self.powercycle_happening:
            self.logger.put("Quake radio was not powercycled.")
        else:
            self.logger.put("Comms blackout caused a powercycle of Quake.")
`
On master, this doesn't look like it's a hard verification of expected behavior? Shouldn't we be throwing a test case failure if we're power cycling when we're not expecting to?

tanishqaggarwal, 2020-09-20:
Fixing this in #498

closed 2020-09-20 by tanishqaggarwal