pathfinder-for-autonomous-navigation/FlightSoftware #495: Check that subsystems respond gracefully to power loss conditions

tanishqaggarwal, 2020-09-09:
Power cycles and power dropouts necessitate more careful tracking of these bugs.
- Power cycling creates a delay between an output going off and coming back online

For example:
- QuakeManager needs to check the "powered" flag prior to doing comms
- Do ADCS, prop, and Piksi have this problem? #487 tracks an ADCS issue where it takes some time to ramp up. But the other issues don't have associated tickets.

shihaocao, 2020-09-20:
@piksi 

Piksi should be robust to power loss at anytime and resumption at any time. With regard to getting piksi CT up to date, this ticket is relevant: #98 
#98

tanishqaggarwal, 2020-09-20:
Prop: we need to add is_functional checks before can_enter for every state, not just IDLE.
Quake: We should have the QuakeFaultHandler handle the QuakeManager state before and after the powercycle.

shihaocao, 2020-09-20:
ADCS:

Modifications:
Add loop(); once before the interrupts in main.cpp in the setup() function.

For each of the following cases, check that the ADCS works as intended before and after. Take care to watch for any DCDC + motor interplay as we saw some before. We initiate/execute each of these cases at the software level using the same state machines.

- Whole satellite reboot
- ADCS reboot only (nevermind since we dont ever autonomously reboot the ADCS, if we did, we'd probably just reboot the whole sat)

^Probably a manual test + documentation of behavior

PR #507 (2020-10-04): Power loss conditions
# Power Loss Conditions

Fixes #495 

### Summary of changes
- Update Quake to check for power cycling command before leaving wait and update QuakeFaultHandler to ensure power cycles only start when in wait 
- Update Prop to check for is_functional() before state changes
- Add loop() to ADCS to populate state fields before I2C begins 
- Add unit tests for the above changes

### Testing

Added unit tests for the changes added and ensured old unit tests pass correctly with the modifications.

### Documentation Evidence
Overall small modifications that are documented in line [... cut]
closed 2020-10-04 by afoarce