pathfinder-for-autonomous-navigation/FlightSoftware #520: Does serialization of integers work if the integer is set to the maximum serializable value?

tanishqaggarwal, 2020-10-18:
Some exploration is needed around the edge cases. Currently being done by @athena255

athena255, 2020-10-18:
Max is inclusive for unsigned and signed integers. 

This can be verified by test_serializer.cpp. On line 235 and 236, both serializers are initialized with min=10, max=10, compress=10. On lines 238 and 239, we see that the serialization of a 10 results in an output value of 10. 

Additionally, the IntegerSerializer constructors have an assert (min <= max). If max were exclusive, then the assert should have been assert (min < max). 

General Comments:
It is dangerous to use subtraction on unsigned numbers if we are not certain of the minimum value that the unsigned number can take. For example, in SerializerTypes.inl, on line 411, we have `quat_sz - 3*quat_component_sz`, but there does not seem to be any checks on the range of `quat_sz` and 'quat_component_sz`. So this constant could take on a very large value if `quat_component_sz` is too large. 

It is not clear to me what the purpose of compressed_size is. This parameter is documented as "Size of the state field when being sent in a downlink". It is unclear to me why on line 272 of test_serializers.cpp, the output of serializing a 3 is a 2 yet on line 273, the output of serializing a 4 is a 4.

tanishqaggarwal, 2020-10-20:
Sweet, thanks for checking this out. Looks like we're good on this front.

closed 2020-10-20 by tanishqaggarwal