pathfinder-for-autonomous-navigation/FlightSoftware #547: Gomspace PPT mode is not persistent across spacecraft reboots

tanishqaggarwal, 2020-11-13:
The Gomspace only respects the _hardware config_ setting of the PPT mode when the spacecraft is off. Changes made via the I2C interface do not persist in the Gomspace's memory after the spacecraft turns off. This poses a problem because we absolutely need the Gomspace PPT mode to be "auto" when in flight even when the spacecraft is off, and for it to be "fixed" when on the ground when the spacecraft is off (so that we can charge via the barrel-jack connector.)

I think the right fix to this would be to create code that can use the I2C interface to change the Gomspace config settings for PPT mode. It doesn't necessarily need to be a part of the main Flight Software (it could be a separate, independent Arduino script), but it could use the existing configuration-changing functions inside the Gomspace driver to accomplish this.

tanishqaggarwal, 2020-12-15:
A side effect of this work is that we also have to modify the ResetForFlight and PrepForCharging cases to edit the Gomspace configs, rather than setting the surface-level PPT mode.

kylekrol, 2021-02-10:
This is actually now a non-issue. The hardware default PPT mode was set to MPPT over gosh. This means whenever booted the spacecraft will be in MPPT and we'll only charge it when the GOMSpace itself is off - i.e. the entire satellite is shutdown.

closed 2021-02-10 by kylekrol

tanishqaggarwal, 2021-02-10:
You might want to track removing PrepForCharging from the procedures.