pathfinder-for-autonomous-navigation/FlightSoftware #572: Cannot read solar panel charging currents when testing with flight units

tanishqaggarwal, 2020-12-09:
We were not able to get charging currents for the solar panels when testing with the flight units. Unclear if we're just reading the wrong state fields (unlikely), or if some mode setting is not correct. We did see the vboost voltages change.

May want to reach out to Gomspace to ask them?

kylekrol, 2021-01-23:
If we can verify that the battery actually charges that would be sufficient to close this ticket.

shihaocao, 2021-02-10:
LEADER DATA:
[image]

shihaocao, 2021-02-10:
FOLLOWER DATA:
[image]

shihaocao, 2021-02-10:
Since all three curin's show response, that shows us that all 3 boost converters have the ability to draw power, on both satellites.

Further, since cursun is positive at some point through both tests, that gives us confidence that the satellites will be able to charge in space while the satellite is on.

closed 2021-02-10 by kylekrol