pathfinder-for-autonomous-navigation/FlightSoftware #573: Verify that all hardware checkout cases are still OK

tanishqaggarwal, 2020-12-09:
We should run them in HOOTL, EDU HITL, and finally both flight units.

18kudlugim, 2021-02-20:
[image]

18kudlugim, 2021-02-20:
DCDC Tests 2 and 4 had issues with resetting the pin. We expect ADCS pin to be true after reset but this doesn't seem to be happening.

18kudlugim, 2021-02-20:
[image]

Gomspace power cycling issues in hitl. Do we need all output channels to be off at the same time? #485

18kudlugim, 2021-02-20:
PiksiCheckoutCase and PropStateMachineCase look good in HITL. Haven't yet run ADCSCheckoutCase and SpinMotorsCase.

knk38, 2021-03-03:
We got rid of SpinMotorsCase and verified that ADCSCheckoutCase works. Additionally we ran all other checkout cases in HITL. Fixes for Gomspace checkout and other cases that threw faults are in pr #647

closed 2021-03-03 by knk38