pathfinder-for-autonomous-navigation/FlightSoftware #582: log function implementation fix causes DownlinkProducer unit tests to fail

tanishqaggarwal, 2020-12-11:
The DownlinkProducer test data is dependent on an incorrect log function implementation, so it fails when the implementaiton is corrected. See failure log on #581

tanishqaggarwal, 2020-12-15:
I forget if this bug causes log to return 1 minus the true value or 1 plus the true value. If it's 1 minus, it's a problem.

tanishqaggarwal, 2021-01-03:
Will address as part of downlink auto-testing, #499

kylekrol, 2021-04-19:
Hey @tanishqaggarwal did we end up cleaning this up?

tanishqaggarwal, 2021-04-19:
No :(

tanishqaggarwal, 2021-04-19:
There is a PR that fully addresses this issue but is semi-working on others: https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/pull/596/files

kylekrol, 2021-04-21:
What's the implication of the bug? Is it critical?

kylekrol, 2021-04-21:
Maybe we can talk about this briefly tomorrow as well

tanishqaggarwal, 2021-04-22:
@kylekrol It could be critical, it affects the bitsizes of all downlinked fields.

The fix itself is quick (just look at the change in SerializerTypes.inl in the above PR). The problem is that we can't drop it in because it breaks the hardcoded downlink unit tests we currently have. Plus, making the change would require re-verification that any telemetry we're getting is correct. I have high confidence that it is (one purpose of the PR above is to show, on an automated test set, that the log2i fix works.) But I was unable to merge in the above PR because of segfaulting issues

tanishqaggarwal, 2021-04-22:
The easiest way forward would be to make the change in SerializerTypes.inl, delete hardcoded unit tests for downlinks, and then do a manual verification that telemetry stuff still works. I hope this is easy enough, given that we've streamlined end-to-end testing quite a bit? I haven't actually attended a recent telemetry test so I'm not sure how easy this is.

shihaocao, 2021-05-27:
Looks like serializers are working so "whatever" as things are still serializing properly.

closed 2021-05-27 by shihaocao