pathfinder-for-autonomous-navigation/FlightSoftware #591: Prop/DCDC power loss

18kudlugim, 2020-12-17:
We need a PTest case that tests the interaction between DCDC and the Prop System. The PropController can only change state if the DCDC SpikeDock pin is set to HIGH. We want to ensure that this behavior is stable.

shihaocao, 2021-01-07:
#609 Does this change things?

athena255, 2021-01-08:
Yes, Prop now sets DCDC pin to HIGH before it attempts to open valves. The ptest case should check for instances when Prop has already set the DCDC pin to HIGH, but test behavior where the pin is suddenly LOW. The states where this matters is: Await_Pressurizing, Await_Firing, Firing, and Venting. 

A unit test has been written to check this behavior when the Prop is in Await_Pressurizing. See `test_prop_controller.cpp`: test_not_enough_cycles_for_dcdc_high, test_enough_cycles_for_dcdc_high, test_require_consecutive_dcdc_high.

shihaocao, 2021-04-01:
@18kudlugim is this cleared per latest batch of PRs?