pathfinder-for-autonomous-navigation/FlightSoftware #604: Ensure no memory leaks of FlightSoftware

tanishqaggarwal, 2021-01-03:
We should run FSW for a long, long time and check if memory usage keeps increasing.

samuelqian01, 2021-04-17:
[image]
[image]

No memory leaks after many many many hours of running.

angelaloh09, 2021-05-08:
3 Hr Quake Test shows no memory leaks:
[image]

angelaloh09, 2021-05-08:
70 min EduSat shows no memory leaks and normal cycling: (detumble state)

No Memory Leaks:
[image]

Cycling: 
[image]

closed 2021-05-12 by angelaloh09