pathfinder-for-autonomous-navigation/FlightSoftware #617: [PTest] - Automatic telem / dbtelem

shihaocao, 2021-01-17:
# Why
We need a system to mimic grabbing data out of a simulated satellite instance that does not have a quake that still uses flows/serializers to generate binary telemetry, and send it to GSW.

# What we have right now
Currently USB session has telem, and dbtelem which can be called from the ptest terminal.

# What we need to do
- We need to expose those command-line accessible commands to commands that are automatically called from a PTest case
- Need a simple configuration parameter that sets the frequency of downlinks, say 10% to represent the idea that only 10% of our downlinks will be received.

Duncan-McD, 2021-03-15:
Pretty sure this is completely done?

Duncan-McD, 2021-03-15:
Only thing we might want to consider is to not have autotelem spam the cmdclient with the hex? makes it pry difficult to enter any commands

closed 2021-04-01 by shihaocao