pathfinder-for-autonomous-navigation/FlightSoftware #624: Verification of control cycle times

knk38, 2021-02-05:
2 main issues:
-We need to verify control cycle times to make sure that they are not consistently late or displaying "cascading" lateness (where one task is late and causes all subsequent tasks to also be late)
-We need to make sure that the num_lates field in the timed control task is accurately counting the number of late tasks.

Some issues were observed when reading multiple gomspace state fields over a period of two minutes which causes an excessive number of tasks to be late in HITL. In HOOTL, the quantity of late tasks is lower but still consistently increasing which is a problem. Shown Below:

HITL:
[image]

HOOTL:
[image]

These times were found after running the gomspace logger ptest case which just reads a collection of gomspace fields for two minutes. However, when running the Magnetorquer test case in HOOTL similar behavior was found, the difference is the late tasks weren't increasing, which is due to the nature of the test. 
[image]

knk38, 2021-03-27:
also related to #532

shihaocao, 2021-04-21:
- Verify that if Flight Software is faster than 170 ms that we lower bound at 170 ms
- Add verifications to hardware checkout cases that every control task executes at 170ms
- Allocate debug task as much as we can up to letting everything sum to 170 ms
- Allocate for quake worst case (40k micros)? + gather more data

PR #734 (2021-05-14): Control Task Timing Fixes
# Control Task Timing Fixes

Fixes #624 

### Summary of changes
- Added a new field that represents the duration of a control task. (timing." control task name ".duration)
- Changed offset representation to update the time after the end of each task rather than using a raw offset from the start of the control cycle.
- Shifted allotted time for each control task 
- Added a new field that represents the duration of a control cycle (pan.cc_duration)

### List of Timing Values Spreadsheet
https://docs.google.com/spreadsheets/d/1jCz0EPqmQYzXp4ZEQNZETDVkG-WAafRNskDdm0EUKvc [... cut]
closed 2021-05-14 by knk38