pathfinder-for-autonomous-navigation/FlightSoftware #691: Telemetry and Commanding Infrastructure Verification

shihaocao, 2021-03-30:
Need to come up with a procedure to connect to satellite and how to send commands (especially for people without much experience in telemetry)
Come up with mission procedures: what should we do if some fault happens or something goes wrong?
Send commands and check that appropriate states on the satellite change/proper response occurs
i.e if we see a fault occur, and we send a command to suppress the fault, then we should get downlinked telemetry showing that fault is suppressed
Critical command fields: resetting fields (i.e gomspace), fault suppressing commands, etc.
Check that we can get all the telemetry in the flows
Check that we can reorder flows of telemetry

shihaocao, 2021-04-01:
#598 parallel track

kylekrol, 2021-04-01:
Related to #597 and #599.