pathfinder-for-autonomous-navigation/FlightSoftware #695: Day-In-The-Life Test

kylekrol, 2021-04-01:
We want to run flight software for an extended period on time (on the order of a day or more) to verify flight software is actually stable enough - i.e. no subtle memory leaks and execution time remains constant over time.

This ideally would be run on an actual Teensy but we could think about running a long duration HOOTL case as well if we monitor memory usage.

Relates to #604 and #624.

We may have issues with #557.

shihaocao, 2021-04-06:
Want to track memory usage over Day in the Life, as well as cycles passed (indirect verification of timing)

angelaloh09, 2021-05-08:
Goals of Ditl case were to fsw can run for long period of time, check for regular control cycle times & to ensure no memory leaks. 
- Verification of control cycle times with EDU 70 min. test attached in #624
- Verification of no memory leak plots/log screenshots for HOOTL Logging, 3HR Quake, and 70 min. Edusat attached in #604

closed 2021-05-12 by angelaloh09