pathfinder-for-autonomous-navigation/FlightSoftware #702: Hard Iron Magnetometer Offsets

kylekrol, 2021-04-05:
Our magnetometer readings have constant offsets induced by the electronics and magnets already present on the vehicle. This must be accounted for prior to passing magnetometer data to the attitude estimator.

kylekrol, 2021-05-19:
[Follower_MagRoll.zip](https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/files/6509623/Follower_MagRoll.zip)
[Leader_MagRoll.zip](https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/files/6509624/Leader_MagRoll.zip)

kylekrol, 2021-05-19:
For the leader:
[image]
[image]

kylekrol, 2021-08-07:
For the follower:
[image]
[image]

kylekrol, 2021-08-07:
[logs_leader.csv](https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/files/6950021/logs_leader.csv)
[logs_follower.csv](https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/files/6950022/logs_follower.csv)

kylekrol, 2021-08-08:
For the leader the mag1 offset is given as:
```
c =

   1.0e-04 *

   -0.5108    0.1698   -0.3657
```

kylekrol, 2021-08-08:
For the follower the mag1 offset is given as:
```
c =

   1.0e-04 *

    0.2716    0.4158   -0.6230
```

kylekrol, 2021-08-08:
```
%% extract data
clc; clear all; close all; format compact
for j = 1:2
    if j == 1
        filename = 'logs_leader';
    else
         filename = 'logs_follower';
    end
    D = readtable([filename '.csv']);
    D = table2cell(D);
    mag1 = []; %x,y,z Teslas
    mag2 = []; %x,y,z Teslas
    c = 1; %counter
    for i = 1:length(D)-2
        if strcmp(D{i+0,2},'adcs_monitor.mag1_vec') == 1 && ...
           strcmp(D{i+1,2},'adcs_monitor.mag2_vec') == 1

            mag1(c,1) = D{i+0,3};
            mag1(c,2) = D{i+0,4};
            mag1(c,3) = D{i+0,5};

            mag2(c,1) = D{i+1,3};
            mag2(c,2) = D{i+1,4};
            mag2(c,3) = D{i+1,5};

            c = c + 1;
        end
    end
    save([filename '.mat'])
end

%% Load the leader or follower

%load('logs_leader.mat')
load('logs_follower.mat')

%% Plot some point clouds (didn't do good enough spins for a sphere)

figure

subplot(1,2,1)
plot(mag1(:,1),'-o'); hold on; grid on
plot(mag1(:,2),'-o')
plot(mag1(:,3),'-o')

subplot(1,2,2)
plot(mag2(:,1),'-o'); hold on; grid on
plot(mag2(:,2),'-o')
plot(mag2(:,3),'-o')

figure

subplot(1,2,1)
plot3(mag1(:,1),mag1(:,2),mag1(:,3),'.')
axis equal

subplot(1,2,2)
plot3(mag2(:,1),mag2(:,2),mag2(:,3),'.')
axis equal

%% Assume mag2 has no offset and attempt to line up mag1

c = mean(mag1 - mag2)

%% Plot calibrated data

mag1_calib = mag1 - c;

figure

plot(mag1_calib(:,1),'-o'); hold on; grid on
plot(mag1_calib(:,2),'-o')
plot(mag1_calib(:,3),'-o')

figure
plot(mag1_calib(:,1) - mag2(:,1)); hold on; grid on
plot(mag1_calib(:,2) - mag2(:,2))
plot(mag1_calib(:,3) - mag2(:,3))
```

kylekrol, 2021-08-08:
TL;DR; Magnetometer one is too close to the wheels and it seems like this imparts a somewhat large dynamics offset into the readings. Secondly, the point cloud sweeped out didn't end up being spherical so we're just assuming magnetometer two doesn't have a dynamics offset.

tanishqaggarwal, 2021-08-15:
@kylekrol are the units Teslas? Just checking to make sure that Earth's magnetic field doesn't matter in the above data collection.

closed 2021-08-16 by kylekrol

kylekrol, 2021-08-21:
> @kylekrol are the units Teslas? Just checking to make sure that Earth's magnetic field doesn't matter in the above data collection.

They are in Tesla's. Discussed offline, however, the idea is that we tumble the satellites in a circle so the Earth's magnetic field rotates throughout SO3 in the body frame and we therefore decouple Earth's magnetic field from the actual offsets due to magnets in the body frame.

PR #794 (2021-08-16): Magnetometer Calibration
# Magnetometer Calibration

Closes #701.
Closes #702.