pathfinder-for-autonomous-navigation/FlightSoftware #803: Telemetry Updates

knk38, 2021-08-31:
- We want to make our telemetry “smarter” in that in the case of a fault we get exactly the data we need rather than everything (this would reduce overhead in sifting through flows to find what we need)
- Additionally it would be nice to send a more concise list of necessary telemetry at each stage of the mission
- We need to figure out how we can implement this system(would probably require a new feature: maybe 100 ish lines of code?)
- Test it after implementation

angelaloh09, 2021-09-25:
Compiled code with: fsw_native_leader_autotelem and fsw_native_follower_autotelem

Ran ptest: python -m ptest runsim -c ptest/configs/hootl_hootl_autotelem.json -t DualSatStandbyCase

Ptest error-ed out while waiting for boot utility waiting to reach initial states (shown in image below): 

[image]

Attempted deleting pio/build folder + rebuilt & deleted eeprom.json.

angelaloh09, 2021-09-25:
Next steps: updating downlink parser. Need to check the mission state to change the order of parser (since the flowids have been reordered for different states.)

shihaocao, 2021-09-26:
Question on implementation here: Why are different flow_data.cpp files required?

I imagine that this can be implemented by switching the flow order alone and leaving the contents of the flows the same?

fatimayousuf, 2021-09-26:
> Question on implementation here: Why are different flow_data.cpp files required?
> 
> I imagine that this can be implemented by switching the flow order alone and leaving the contents of the flows the same?

So if you look at the way to reorder flows right now, you have to enter the ids of the flows you want to shift around, not the index of the flow in the flow order. If we keep changing the flows around from mission state to mission state, then you wouldn’t know where the flow you want to move is in the flow order. Idk, I just thought it would easier to define different flow orders for each mission state, but I think we could do it that way too.