pathfinder-for-autonomous-navigation/FlightSoftware #812: Estimator FDIR: Autonomously Reset the Attitude Estimator on Large Estimate Covariance

kylekrol, 2021-09-24:
Per the title.

We should run some `psim` cases with and without sun sensors, baseline the covariance frobenius norm we expect to see, add a large factor of safety (like 1000x or more), and then implement a feature that autonomously resets the attitude estimator if we exceed that for a certain persistence -- kinda like a fault but doesn't affect the state machine as much.

There should be a way to disable this feature from the ground in case it gets out of control and/or our bound is bad.

kylekrol, 2021-09-24:
Similar features could be ticketed for the other estimators -- although I am most worried about the attitude estimator particularly if we aren't using sun vector measurements.

kylekrol, 2021-09-24:
cc: @govindchari

govindchari, 2021-09-25:
Sam currently has the flight software portion of this task done and is waiting on the thresholds from my psim runs. I made some changes to the flight computer model in psim and am working on getting baseline covariances.