pathfinder-for-autonomous-navigation/FlightSoftware #813: Bootcount is overwritten by EEPROM

shihaocao, 2021-09-25:
Two things to fix:
Lets loosen the restriction on MissionManager

Second lets change the computation of bootcount and its interaction with EEPROM so the statefield is still meaningful

shihaocao, 2021-09-29:
Post PR Verification:
(venv) shihao@shihao-T490V2:~/Code/PAN/FlightSoftware$ python -m ptest runsim -c ptest/configs/hitl.json -t EmptyCase
 _______     _       ____  _____                                
|_   __ \   / \     |_   \|_   _|                            
  | |__) | / _ \      |   \ | |                               
  |  ___/ / ___ \     | |\ \| |                              
 _| |_  _/ /   \ \_  _| |_\   |_                             
|_____||____| |____||_____|\____|                              
                                                                
Pathfinder for Autonomous Navigation                            
Space Systems Design Studio, Cornell University       
Console Software | Version 1.0

Opened connection to FlightController.
Running mission testcase EmptyCase.
[TESTCASE] Starting testcase.
[TESTCASE] Suppressing faults.
Turning off fault_handler
Turning off piksi_fh
Turning off qfh
Supressing gomspace.low_batt
Supressing gomspace.get_hk.base
Supressing prop.overpressured
Supressing prop.tank2_temp_high
Supressing prop.tank1_temp_high
Supressing adcs_monitor.functional_fault
Supressing adcs_monitor.wheel1_fault
Supressing adcs_monitor.wheel2_fault
Supressing adcs_monitor.wheel3_fault
Supressing adcs_monitor.wheel_pot_fault
Supressing attitude_estimator.fault
Supressing piksi_fh.dead
[TESTCASE] Auto-dbtelem INACTIVE
[TESTCASE] Scrape-uplink ACTIVE
[TESTCASE] Boot utility waiting to reach initial state.
Beginning console.
Type "help" for a list of commands.
NOTE: You are currently connected to the FlightController.
> [TESTCASE] Finished testcase.

> rs pan.bootcount
1                                               (Completed in 5596 us)
> exit
Exiting command line.
Stopping binary monitor thread...
Stopping uplink console...
Terminating 0 radio connection(s)...
Terminating 1 device connection(s)...
 - Terminating console connection to and saving logging/telemetry data for FlightController.
 - Finished terminating for FlightController.
(venv) shihao@shihao-T490V2:~/Code/PAN/FlightSoftware$ python -m ptest runsim -c ptest/configs/hitl.json -t Reboot
 _______     _       ____  _____                                
|_   __ \   / \     |_   \|_   _|                            
  | |__) | / _ \      |   \ | |                               
  |  ___/ / ___ \     | |\ \| |                              
 _| |_  _/ /   \ \_  _| |_\   |_                             
|_____||____| |____||_____|\____|                              
                                                                
Pathfinder for Autonomous Navigation                            
Space Systems Design Studio, Cornell University       
Console Software | Version 1.0

Opened connection to FlightController.
Running mission testcase Reboot.
[TESTCASE] Starting testcase.
[TESTCASE] Suppressing faults.
Turning off fault_handler
Turning off piksi_fh
Turning off qfh
Supressing gomspace.low_batt
[... 84 lines cut ...]
Supressing adcs_monitor.functional_fault
Supressing adcs_monitor.wheel1_fault
Supressing adcs_monitor.wheel2_fault
Supressing adcs_monitor.wheel3_fault
Supressing adcs_monitor.wheel_pot_fault
Supressing attitude_estimator.fault
Supressing piksi_fh.dead
[TESTCASE] Auto-dbtelem INACTIVE
[TESTCASE] Scrape-uplink ACTIVE
[TESTCASE] Boot utility waiting to reach initial state.
Beginning console.
Type "help" for a list of commands.
NOTE: You are currently connected to the FlightController.
> [TESTCASE] Finished testcase.

> rs pan.bootcount
2                                               (Completed in 5172 us)
> Exiting command line.
Stopping binary monitor thread...
Stopping uplink console...
Terminating 0 radio connection(s)...
Terminating 1 device connection(s)...
 - Terminating console connection to and saving logging/telemetry data for FlightController.
 - Finished terminating for FlightController.
(venv) shihao@shihao-T490V2:~/Code/PAN/FlightSoftware$

shihaocao, 2021-09-29:
See that in the above log, an Empty Case was run for the first time where bootcount was 1, then it was reboot, and then the next time the Empty Case shows bootcount as == 2.

knk38, 2021-09-29:
Okay, so looks like we can close this issue given we see bootcount change?