pathfinder-for-autonomous-navigation/FlightSoftware #825: Standby Wheels Spin out of Control

shihaocao, 2021-10-02:
In HOOTL and on HITL, running a SingleSatStandbyCase has the wheels slowly increasing their wheel speeds over time.

This can be seen by plotting truth.leader.attitude.w. This is also reflected in HITL as the wheel speeds grow over time.

This should not be the case as in Standby, the wheels should spin up for a brief moment to slew towards the target pointing vector and then slow to a steady-state slow rotation to keep the spacecraft from spinning. They should not be continuously increasing speed over time up to the 50 to 60 radians per second level. This does not make sense.

shihaocao, 2021-10-02:
[image]

Is the leader attitude L climbing because psim includes wheel speeds? But either way, wheel speeds should not be climbing.

shihaocao, 2021-10-02:
Wheel speeds climbing a ton, wheel speed Y-axis seems to follow attitude estimator L body estimates...

shihaocao, 2021-10-02:
All seems to be well and good here, but I don't understand how wheels spinning up faster and faster can still have the q_body_eci stay tame.

[image]

**something must be broken with psim**

shihaocao, 2021-10-02:
Nominal timings: 
[image]

Cycle time seems to be good enough

kylekrol, 2021-10-06:
@shihaocao
@govindchari 

Okay, so too me it seems like somehow we're loosing/gaining angular momentum in a way that we shouldn't be able to. Basically, we continue to command the wheels to try and stabalize ourselves; yet, the wheels continue to store more and more angular momentum. Either we changed `psim` and the dynamics are no longer correct or our handling of MTR commands is wack -- and causing us to just build up more and more angular momentum -- or perhaps the step size right now is less than ideal.

Attaching some plots below, but you can see we're "stable" until the wheels have too much stored angular momentum and the nonlinear effects would then be too much for us to handle -- presumably.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

kylekrol, 2021-10-06:
So things seem to be fine at this commit:

* https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/commit/8521bbae80ebd68d662a99cbcc60dc29bf076a7a

[image]

kylekrol, 2021-10-06:
Between that last commit and this guy:

* https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/commit/10d940cfd96f1fe7aea56f4db57d1ee239a99396

kylekrol, 2021-10-06:
I think this is the problem child: https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/commit/ee2f07861826e3597a8ad8c3fcb060a2770c0030

kylekrol, 2021-10-06:
I think this has been an issue the whole time actually and just hidden by a high detumble threshold. Checkout these results from `testing/wheel-speed` which is `master` with old detumble thresholds:

[image]
[image]
[image]

The bottom plot of `w` is the right half of the first plot just zoomed in and it also shows the unstable behavior. Checkout the wheel speeds as well!

kylekrol, 2021-10-06:
I suspect the MTR controller given I don't this this unstable behavior happens in `psim`? Although, I guess we don't run our controller there so...

kylekrol, 2021-10-06:
MTR controller is borked; zeroing out the command sent to `psim` gives us this:

[image]

kylekrol, 2021-10-06:
```
# Simulate the reaction wheels speed read
sensors["adcs_monitor.rwa_speed_rd"] = list(self.__sim[f"truth.{satellite}.wheels.w"])
```

shihaocao, 2021-10-07:
with the described changes, looks good now:
[image]
[image]

MTR cmds are 0:
[image]