pathfinder-for-autonomous-navigation/FlightSoftware #831: GPS Time Deserializer Not Working

shihaocao, 2021-10-15:
[image]

shihaocao, 2021-10-15:
git checkout gpstime-test

ws a time.gps, notice that after telem dump + parsetelem that all odd numbered wn don't parse correctly

shihaocao, 2021-10-17:
[image]
with the 2nd flow and 3rd flow swapped positions, we note that velocity also happens to be negative when it should be positive...

This output was generated with the new FlowInpsect ptest case

afoarce, 2021-10-31:
[image]
In printing snapshots over time, it appears that the 'cursed' bit flips when the packet splits based on testing on branch `cursed-testing`

afoarce, 2021-10-31:
I believe it is this [line](https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/blob/5ed3ece09182b82725f0028a2045cc33dd82a8f3/src/fsw/FCCode/DownlinkProducer.cpp#L100) that clears the bit due to a miscalculation on [line] (https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/blob/5ed3ece09182b82725f0028a2045cc33dd82a8f3/src/fsw/FCCode/DownlinkProducer.cpp#L99). When splitting packets, the downlink_frame_offset/70 gives 8 (bytes) and so the 64th bit is cleared instead of the first bit after 70 bytes. The packet header is still added successfully because snapshot is initialized to all 0s and the offset is incremented, so the 561st bit remains 0. Testing with the modify_bit line commented out shows no cleared bit. 

Unfortunately, this means the combination of changing the serializer size for bootcount and adding the cursed bit in #838 moved the issue but did not mitigate it. Further testing needed to quantify the effects on the accuracy of `time.gps`. 

@shihaocao @tanishqaggarwal @kylekrol

shihaocao, 2021-10-31:
wow that's crazy

shihaocao, 2021-11-06:
The bit that is being set is now in pan.cycle_no, therefore we are fine.

shihaocao, 2021-11-06:
[image]

shihaocao, 2021-11-06:
[image]
In the second half of the packet, the rwa_torque_cmd bit will be set to zero as it takes up 53 - 65

knk38, 2021-11-06:
Resolution is to turn off cursed flows, and only toggle on these flows:
- 14, 15
- 17 - 19 (inclusive)
- 25, 26

shihaocao, 2021-11-06:
Or we change the downlink parser code to go as follows:
if a packet claims it is the first packet in a new frame, parse it immediately, check its flow_id values in the expected positions if they don't comprise the flow_id values we expect in the first packet of a new frame, conclude that it must be a packet a part of the previous frame.