pathfinder-for-autonomous-navigation/FlightSoftware #852: Uplink command log

tanishqaggarwal, 2022-01-17:
Goal:
Put every uplink sent to spacecraft as a new comment and post rationale for each packet sent.

shihaocao, 2022-01-17:
Command:
pan state to manual
because belief that safehold could be bugged,
manual is basically safehold, but without fault response, but this is okay because ADCS isn't working anyway

downlink.toggle_id to 11,
because turn on gomspace boot cause flow

radio.max_wait to 10420 (approx 30 mins),
because want to limit quake power usage without limiting our transmission rate too much. we're rebooting every 38 mins anyway, longer durations likely don't mean much.

havt_disable18 disables the gyro heaters. This is a power-saving measure. This also assumes i2c comms with the ADCS is working. Noteworthy, if we see the havt disable later report heater as non functional, we know i2c is working with adcs.

adcs.state to 1 to ensure ADCS stays passive.

gomspace.piksi_off to turn off piksi as a power saving measure.
It's broken anyways, and we're not using it.

[
  {
    "field": "pan.state",
    "value": 11
  },
  {
    "field": "downlink.toggle_id",
    "value": 11
  },
  {
    "field": "radio.max_wait",
    "value": 10420
  },
  {
    "field": "adcs_cmd.havt_disable18",
    "value": true
  },
  {
    "field": "adcs.state",
    "value": 1
  },
  {
    "field": "gomspace.piksi_off",
    "value": true
  }
]

Duncan-McD, 2022-01-18:
**MTMSN 15**

This is one more uplink of PowerSaving2 to the leader. The rational for each of the fields in this command remains the same as MTMSN 12,13, and 14.

We are sending one more of these uplinks as we saw a promising increase in voltage in the most recent downlink and we want to see if the trend remains.

`
[
  {
    "field": "pan.state",
    "value": 11
  },
  {
    "field": "downlink.toggle_id",
    "value": 11
  },
  {
    "field": "radio.max_wait",
    "value": 10420
  },
  {
    "field": "adcs_cmd.havt_disable18",
    "value": true
  },
  {
    "field": "adcs.state",
    "value": 1
  },
  {
    "field": "gomspace.piksi_off",
    "value": true
  }
]

`

Queue'd **1** time at 7:56 PM ET on 1/17/2022

shihaocao, 2022-01-18:
[
  {
    "field": "pan.state",
    "value": 11
  },
  {
    "field": "downlink.toggle_id",
    "value": 11
  },
  {
    "field": "radio.max_wait",
    "value": 10420
  },
  {
    "field": "radio.max_transceive",
    "value": 10
  },
  {
    "field": "adcs_cmd.havt_disable18",
    "value": true
  },
  {
    "field": "adcs.state",
    "value": 1
  },
  {
    "field": "gomspace.piksi_off",
    "value": true
  },
  {
    "field": "radio.max_transceive",
    "value": 10
  },
  {
    "field": "dcdc.disable_cmd",
    "value": true
  },
  {
    "field": "attitude_estimator.ignore_sun_vectors",
    "value": false
  },
]

**PowerSaving3**

Introduces two new power-saving measures
max_transceive to 10 cycles, to attempt to transceive 50 times less,
dcdc.disable_cmd to true, in order to ensure DCDCs are off if they are not already off

Toggles attitude_estimator.ignore_sun_vectors the other way to confirm application of commands.

tanishqaggarwal, 2022-01-18:
**MTMSN 17**
Re-queuing PowerSaving3. Although the battery level observed on the previous downlink was low (6680), this may not be due to the power-saving measures but instead due to eclipse.

We don't have enough data yet to see if the previous uplink reduced the bootcount rate.

tanishqaggarwal, 2022-01-18:
**MTMSN 18**
Per our discussion earlier this evening, I am re-queuing PowerSaving3. The battery voltage increased to 7080 mV, and we agreed to continue queueing PowerSaving3 for as long as there is no apparent worsening of battery health over time.

Queued 1 time at 2:02 AM ET on 1/18/2022

fatimayousuf, 2022-01-18:
**MTMSN 22**
Following up on the previous log entry, I re-sent PowerSaving3.

The leader's vbatt has been oscillating between ~6700 and 7100. The boot count rate seems to have stayed the same

Queued at 10:18 AM PST/1:18 EST on 1/18/2022

stewartaslan, 2022-01-19:
**MTMSN 23**

> IMEI: 300234030726020
> Time: Wed Jan 19 03:33:56 2022
> 
> Attachment Filename: http_uplink300234030726020.sbd
> Attachment Size: 15
> 
> The MTMSN is 23, and the message is number 1 in the queue.

PowerSaving3 queued with `attitude_estimator.ignore_sun_vector` set to `true` to show that Leader received the command.

govindchari, 2022-01-19:
MTMSN 24

PowerSaving3 queued at 11:30am EST on January 19th

shihaocao, 2022-01-19:
MTMSN 24

PowerSaving3 queued at 3:10 EST on 01/19. Set ignore sun vector to `false`

stewartaslan, 2022-01-20:
MTMSN 27

PowerSaving3 queued 7:42pm, ignore sun vector to true.

stewartaslan, 2022-01-20:
**MTMSN 28**

PowerSaving3 queued 8:51pm pacific time, ignore sun vector set to false

stewartaslan, 2022-01-20:
**MTMSN 29**

PowerSaving3 queued at 9:36pm pacific time, ignore sun vector set to true

stewartaslan, 2022-01-20:
**MTMSN 30**

PowerSaving3 queued at 12:01am pacific time, ignore sun vector set to false

stewartaslan, 2022-01-20:
**MTMSN 31**

PowerSaving3 queued at 12:55am pacific time, ignore sun vector set to false. This should have been true but was sent as false by accident, so we won't be able to tell from this flag whether this command has been applied.

fatimayousuf, 2022-01-21:
**MTMSN 32**

PowerSaving3 uplinked to leader at 8:34pm EST 1/20/22 with the following additions:
  {
    "field": "downlink.toggle_id",
    "value": 36
  },
  {
    "field": "downlink.shift_id1",
    "value": 36
  },
  {
    "field": "downlink.shift_id2",
    "value": 2
  }
This would help us see what gomspace.wdt_i2c_time_left is.

kylekrol, 2022-01-21:
MTSN 33

```
[
  {
    "field": "radio.max_wait",
    "value": 10420
  },
  {
    "field": "radio.max_transceive",
    "value": 10
  },
  {
    "field": "adcs_cmd.havt_disable18",
    "value": true
  },
  {
    "field": "adcs.state",
    "value": 1
  },
  {
    "field": "gomspace.piksi_off",
    "value": true
  },
  {
    "field": "dcdc.disable_cmd",
    "value": true
  },
  {
    "field": "attitude_estimator.ignore_sun_vectors",
    "value": false
  },
  {
    "field": "downlink.toggle_id",
    "value": 36
  },
  {
    "field": "downlink.shift_id1",
    "value": 36
  },
  {
    "field": "downlink.shift_id2",
    "value": 2
  }
]```

shihaocao, 2022-01-21:
MTMSN 34

[
  {
    "field": "radio.max_wait",
    "value": 10420
  },
  {
    "field": "radio.max_transceive",
    "value": 10
  },
  {
    "field": "adcs_cmd.havt_disable18",
    "value": true
  },
  {
    "field": "adcs.state",
    "value": 1
  },
  {
    "field": "gomspace.piksi_off",
    "value": true
  },
  {
    "field": "dcdc.disable_cmd",
    "value": true
  },
  {
    "field": "attitude_estimator.ignore_sun_vectors",
    "value": false
  },
  {
    "field": "downlink.toggle_id",
    "value": 36
  },
  {
    "field": "downlink.shift_id1",
    "value": 36
  },
  {
    "field": "downlink.shift_id2",
    "value": 2
  }
]

Repeated same commands just to get the same power Saving3 and attempt to get the new telemetry points

afoarce, 2022-01-21:
MTMSN 35, PowerSave3 with sun_vectors true

```
[
  {
    "field": "radio.max_wait",
    "value": 10420
  },
  {
    "field": "radio.max_transceive",
    "value": 10
  },
  {
    "field": "adcs_cmd.havt_disable18",
    "value": true
  },
  {
    "field": "adcs.state",
    "value": 1
  },
  {
    "field": "gomspace.piksi_off",
    "value": true
  },
  {
    "field": "dcdc.disable_cmd",
    "value": true
  },
  {
    "field": "attitude_estimator.ignore_sun_vectors",
    "value": true
  },
  {
    "field": "downlink.toggle_id",
    "value": 36
  },
  {
    "field": "downlink.shift_id1",
    "value": 36
  },
  {
    "field": "downlink.shift_id2",
    "value": 2
  }
]
```

afoarce, 2022-01-22:
MTMSN 36, PowerSave3, sun_vectors false
```
[
  {
    "field": "radio.max_wait",
    "value": 10420
  },
  {
    "field": "radio.max_transceive",
    "value": 10
  },
  {
    "field": "adcs_cmd.havt_disable18",
    "value": true
  },
  {
    "field": "adcs.state",
    "value": 1
  },
  {
    "field": "gomspace.piksi_off",
    "value": true
  },
  {
    "field": "dcdc.disable_cmd",
    "value": true
  },
  {
    "field": "attitude_estimator.ignore_sun_vectors",
    "value": false
  },
  {
    "field": "downlink.toggle_id",
    "value": 36
  },
  {
    "field": "downlink.shift_id1",
    "value": 36
  },
  {
    "field": "downlink.shift_id2",
    "value": 2
  }
]
```

jer296, 2022-01-22:
MTMSN 37, PowerSave3, sun_vectors true

`[
  {
    "field": "radio.max_wait",
    "value": 10420
  },
  {
    "field": "radio.max_transceive",
    "value": 10
  },
  {
    "field": "adcs_cmd.havt_disable18",
    "value": true
  },
  {
    "field": "adcs.state",
    "value": 1
  },
  {
    "field": "gomspace.piksi_off",
    "value": true
  },
  {
    "field": "dcdc.disable_cmd",
    "value": true
  },
  {
    "field": "attitude_estimator.ignore_sun_vectors",
    "value": true
  },
  {
    "field": "downlink.toggle_id",
    "value": 36
  },
  {
    "field": "downlink.shift_id1",
    "value": 36
  },
  {
    "field": "downlink.shift_id2",
    "value": 2
  }
]`

samuelqian01, 2022-01-22:
MTMSN 38, requed PowerSaving3 with sun_vectors false

`[
  {
    "field": "radio.max_wait",
    "value": 10420
  },
  {
    "field": "radio.max_transceive",
    "value": 10
  },
  {
    "field": "adcs_cmd.havt_disable18",
    "value": true
  },
  {
    "field": "adcs.state",
    "value": 1
  },
  {
    "field": "gomspace.piksi_off",
    "value": true
  },
  {
    "field": "dcdc.disable_cmd",
    "value": true
  },
  {
    "field": "attitude_estimator.ignore_sun_vectors",
    "value": false
  },
  {
    "field": "downlink.toggle_id",
    "value": 36
  },
  {
    "field": "downlink.shift_id1",
    "value": 36
  },
  {
    "field": "downlink.shift_id2",
    "value": 2
  }
]`

millie-schwartz, 2022-01-23:
MTMSN 39 sun_vectors true

{
    "field": "radio.max_wait",
    "value": 10420
  },
  {
    "field": "radio.max_transceive",
    "value": 10
  },
  {
    "field": "adcs_cmd.havt_disable18",
    "value": true
  },
  {
    "field": "adcs.state",
    "value": 1
  },
  {
    "field": "gomspace.piksi_off",
    "value": true
  },
  {
    "field": "dcdc.disable_cmd",
    "value": true
  },
  {
    "field": "attitude_estimator.ignore_sun_vectors",
    "value": true
  },
  {
    "field": "downlink.toggle_id",
    "value": 36
  },
  {
    "field": "downlink.shift_id1",
    "value": 36
  },
  {
    "field": "downlink.shift_id2",
    "value": 2
  }

tanishqaggarwal, 2022-01-23:
MTMSN 40 ignore_sun_vectors false

```
{
"field": "radio.max_wait",
"value": 10420
},
{
"field": "radio.max_transceive",
"value": 10
},
{
"field": "adcs_cmd.havt_disable18",
"value": true
},
{
"field": "adcs.state",
"value": 1
},
{
"field": "gomspace.piksi_off",
"value": true
},
{
"field": "dcdc.disable_cmd",
"value": true
},
{
"field": "attitude_estimator.ignore_sun_vectors",
"value": false
},
{
"field": "downlink.toggle_id",
"value": 36
},
{
"field": "downlink.shift_id1",
"value": 36
},
{
"field": "downlink.shift_id2",
"value": 2
}
```

shihaocao, 2022-01-23:
{
"field": "radio.max_transceive",
"value": 10
},
{
"field": "adcs_cmd.havt_disable18",
"value": true
},
{
"field": "adcs.state",
"value": 1
},
{
"field": "gomspace.piksi_off",
"value": true
},
{
"field": "dcdc.disable_cmd",
"value": true
},
{
"field": "attitude_estimator.ignore_sun_vectors",
"value": true
},
{
"field": "pan.state",
"value": 11
},
{
"field": "gomspace.power_cycle_output4_cmd",
"value": true
}
]
```

shihaocao, 2022-01-24:
Sent 08:13 ET on 01/24
MTMSN 42
`attitude_estimator.ignore_sun_vectors` : false
  
**PowerSaving3 + ADCS Power Cycle**

Rationale: Repeat the same PowerSavind3 + ADCS Power Cycle command in hopes of seeing a reboot.

shihaocao, 2022-01-24:
Sent 12:13 ET on 01/24
MTMSN 43
`attitude_estimator.ignore_sun_vectors`: `true`

**PowerSaving3 + ADCS Power Cycle**

Rationale: Repeat the same PowerSavind3 + ADCS Power Cycle command in hopes of seeing established communications with the ADCS Computer.

claupiet, 2022-01-24:
Sent 17:28 ET on 01/24
MTMSN 44
attitude_estimator.ignore_sun_vectors: false

PowerSaving3 + ADCS Power Cycle

Rationale: No significant change in values

shihaocao, 2022-01-26:
Sent 7:34 ET on 01/25
MTMSN 45
attitude_estimator.ignore_sun_vectors: true

PowerSaving3 + ADCS Power Cycle

Rationale: " "

afoarce, 2022-01-26:
Sent 2:15 ET, 1/26
MTMSN 47
attitude_estimator.ignore_sun_vectors: true
PowerSaving3 + ADCS Power Cycle

pan-master-chief, 2022-01-27:
Sent 02:26 AM ET on 01/27
MTMSN 48
attitude_estimator.ignore_sun_vectors: false

PowerSaving3 + ADCS Power Cycle

Rationale: " "

shihaocao, 2022-02-05:
Some time on 02/01, changed to this default uplink:
```
[
    {
    "field": "radio.max_wait",
    "value": 10420
    },
    {
    "field": "radio.max_transceive",
    "value": 10
    },
    {
    "field": "adcs_cmd.havt_disable18",
    "value": true
    },
    {
    "field": "adcs.state",
    "value": 1
    },
    {
    "field": "gomspace.piksi_off",
    "value": true
    },
    {
    "field": "dcdc.disable_cmd",
    "value": true
    },
    {
    "field": "gomspace.power_cycle_output4_cmd",
    "value": true
    },
    {
    "field": "downlink.shift_id1",
    "value": 19
    },
    {
    "field": "downlink.shift_id2",
    "value": 2
    }
]
```
PowerSaving3 + ADCS Power Cycle + Mag Debug

Rationale: Moved flow 19 up in order to get more raw magnetometer measurements

afoarce, 2022-02-28:
On 2/18, `adcs_ffa1` was set to the default command for Leader:

```
[
    {
    "field": "adcs_cmd.havt_disable9",
    "value": true
    },
    {
    "field": "pan.state",
    "value": 1  
    },
    {
    "field": "adcs.state",
    "value": 5
    },
    {
    "field": "gomspace.low_batt.suppress",
    "value": true
    },
    {
    "field": "attitude_estimator.fault.base",
    "value": true
    },
    {
    "field": "adcs_monitor.wheel1_fault.suppress",
    "value": true
    },
    {
    "field": "adcs_monitor.wheel2_fault.suppress",
    "value": true
    },
    {
    "field": "adcs_monitor.wheel3_fault.suppress",
    "value": true
    },
    {
    "field": "adcs_monitor.wheel_pot_fault.suppress",
    "value": true
    }
]
```

The goal of this command was to force the satellite into Detumble, hence the `pan.state` and `adcs.state` fields.  To verify if the ADCS bus is functional, the `adcs_cmd.havt_disable9` command is set. Testing and analysis of flight code shows that faults are not updated as necessary and thus ADCS may appear faulted even when functional. Thus, the remaining fields suppress faults that would force the satellite into Safehold and would prevent Detumble.

shihaocao, 2022-04-21:
On 04/17/2022 on Sunday at 12:04 PM ET, sent a command to do 100ms prop firing out of valve 1.

shihaocao, 2022-04-21:
[image]
04/17 approximately correlates to the above red, arrow, not enough of a disturbance in gyro rate was seen, 

Seems safe to queue more prop firings.

shihaocao, 2022-04-21:
On Thursday 04/21/22 at 3:25 PM sent another command for 500ms out of valve 1

shihaocao, 2022-04-22:
On Friday 04/22/22 at 12:47 PM ET sent two more commands (MTMSN 889, 890) for 500ms out of valve1

shihaocao, 2022-04-29:
On Friday 04/29/22 at 1:36 PM sent 3 commands for 900ms out of valve 1 (2700ms total)