pathfinder-for-autonomous-navigation/FlightSoftware #853: Follower Uplink Command Log

shihaocao, 2022-01-17:
[
  {
    "field": "pan.state",
    "value": 11
  },
  {
    "field": "downlink.toggle_id",
    "value": 11
  },
  {
    "field": "radio.max_wait",
    "value": 10420
  },
  {
    "field": "adcs_cmd.havt_disable18",
    "value": true
  },
  {
    "field": "adcs.state",
    "value": 1
  },
  {
    "field": "gomspace.piksi_off",
    "value": true
  }
]

Queue'd once at 2:59 AM ET on 01/17/2022.

jer296, 2022-01-22:
MTMSN 4

Moving to PowerSave3, ignore_sun_vectors true, with ~24 hour radio max_wait for drastic measures

`[
  {
    "field": "radio.max_wait",
    "value": 500000
  },
  {
    "field": "radio.max_transceive",
    "value": 10
  },
  {
    "field": "adcs_cmd.havt_disable18",
    "value": true
  },
  {
    "field": "adcs.state",
    "value": 1
  },
  {
    "field": "gomspace.piksi_off",
    "value": true
  },
  {
    "field": "dcdc.disable_cmd",
    "value": true
  },
  {
    "field": "attitude_estimator.ignore_sun_vectors",
    "value": true
  },
  {
    "field": "downlink.toggle_id",
    "value": 36
  },
  {
    "field": "downlink.shift_id1",
    "value": 36
  },
  {
    "field": "downlink.shift_id2",
    "value": 2
  }
]`

millie-schwartz, 2022-01-23:
MTMSN 5
PS3, sun_vectors false, long wait time

```
[
    {
    "field": "radio.max_wait",
    "value": 500000
  },
  {
    "field": "radio.max_transceive",
    "value": 10
  },
  {
    "field": "adcs_cmd.havt_disable18",
    "value": true
  },
  {
    "field": "adcs.state",
    "value": 1
  },
  {
    "field": "gomspace.piksi_off",
    "value": true
  },
  {
    "field": "dcdc.disable_cmd",
    "value": true
  },
  {
    "field": "attitude_estimator.ignore_sun_vectors",
    "value": false
  },
  {
    "field": "downlink.toggle_id",
    "value": 36
  },
  {
    "field": "downlink.shift_id1",
    "value": 36
  },
  {
    "field": "downlink.shift_id2",
    "value": 2
  }
]
```

shihaocao, 2022-01-26:
Sent 7:34 ET on 01/25
MTMSN 6
attitude_estimator.ignore_sun_vectors: true

PowerSaving3 + ADCS Power Cycle

Rationale: same as leader, see leader log
We want to try the same on the follower, maybe we'll get lucky and see results.

afoarce, 2022-03-06:
Updated goto_follower command to match the leader, rationale on the leader log: https://github.com/pathfinder-for-autonomous-navigation/FlightSoftware/issues/852#issuecomment-1053871234