pathfinder-for-autonomous-navigation/FlightSoftware #863: Downlink Shift id1 and id2 fields not being applied in uplinks

shihaocao, 2022-02-10:


shihaocao, 2022-02-10:
Sample debug output from `bugfix/uplink`:
```
[2022-02-09 22:54:15.181655] (ERROR) UplinkConsumer::execute()

[2022-02-09 22:54:15.181655] (ERROR) Null or length 0

[2022-02-09 22:54:15.235655] (ERROR) DP pre execute
[2022-02-09 22:54:15.235655] (ERROR) shift_flow_ids1_f.get() = 0
[2022-02-09 22:54:15.235655] (ERROR) shift_flow_ids2_f.get() = 0
[2022-02-09 22:54:15.235655] (ERROR) DP post execute

[2022-02-09 22:54:15.327655] Successfully sent telemetry to FlightSoftware.
\x79\x11\x7c\x78\x22\x16\x88\xe6\x44\x52\x0\xc0\x11\x2d\xd\xa\x0
[2022-02-09 22:54:15.347655] (ERROR) UplinkConsumer::execute()

[2022-02-09 22:54:15.347655] (ERROR) Validating

[2022-02-09 22:54:15.347655] (ERROR) packet bytes in validate: 17

[2022-02-09 22:54:15.347655] (ERROR) Validate result 0

[2022-02-09 22:54:15.347655] (ERROR) packet bytes in validate: 17

[2022-02-09 22:54:15.405655] (ERROR) DP pre execute
[2022-02-09 22:54:15.405655] (ERROR) shift_flow_ids1_f.get() = 0
[2022-02-09 22:54:15.405655] (ERROR) shift_flow_ids2_f.get() = 0
[2022-02-09 22:54:15.405655] (ERROR) DP post execute
[2022-02-09 22:54:15.518655] (ERROR) UplinkConsumer::execute()

[2022-02-09 22:54:15.518655] (ERROR) Null or length 0

[2022-02-09 22:54:15.575655] (ERROR) DP pre execute
[2022-02-09 22:54:15.575655] (ERROR) shift_flow_ids1_f.get() = 0
[2022-02-09 22:54:15.575655] (ERROR) shift_flow_ids2_f.get() = 0
[2022-02-09 22:54:15.575655] (ERROR) DP post execute
[2022-02-09 22:54:15.692655] (ERROR) UplinkConsumer::execute()
```

As observed from above, either our uplinks do not contain the `shift` flow fields, or UplinkConsumer isn't Consuming them, or some other FSW is mutating them to be 0 before it gets to DownlinkParser

Duncan-McD, 2022-02-15:
Seems to be fixed by removing the print statements - tested on the bugfix/uplink branch on mac with no print statements, and then tested on the 
862-uplink-swap-fix branch (with the linux error code -9 fix and no print statements) and it works:
[image]